xmm7360/xmm7360-pci #1: Power management is worse than nothing

abrasive, 2020-01-07:
The driver will hang on resume if any queue pairs were open on suspend

abrasive, 2020-01-15:
Removed the dodgy suspend/resume handling in 2066cdb02. Now the thing is just dead when you resume (needs rmmod/modprobe again).

abrasive, 2020-01-15:
Just to be a little clearer:

The modem loses power during suspend; it comes up freshly booted (or booting) on resume.

tgxn, 2020-12-14:
On resume, I am recieving an error:
```
Traceback (most recent call last):
  File "/etc/x1_yoga/xmm7360-pci/rpc/open_xdatachannel.py", line 31, in <module>
    r = rpc.XMMRPC()
  File "/etc/x1_yoga/xmm7360-pci/rpc/rpc.py", line 18, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
Cannot find device "wwan0"
```

I have run rmmod/insmod, and make/make load. To get it working again, a full restart is required.
- X1 Yoga Gen3

flokli, 2021-08-01:
> Removed the dodgy suspend/resume handling in [2066cdb](https://github.com/xmm7360/xmm7360-pci/commit/2066cdb021102fcc73b6cfb029bec5420864f653). Now the thing is just dead when you resume (needs rmmod/modprobe again).

Tried with latest master, can't get it to come back after a suspend, even with a rmmod/modprobe cycle.
`dmesg` shows:

```
[Sun Aug  1 15:04:04 2021] xmm7360 0000:05:00.0: modem is ready
[Sun Aug  1 15:04:05 2021] xmm7360 0000:05:00.0: Could not bring up command ring
[Sun Aug  1 15:04:05 2021] xmm7360: probe of 0000:05:00.0 failed with error -110
```

iMonZ, 2022-02-07:
So every time I suspend the laptop I need to restart it to use this tool again?
Have I understood it correctly?