xmm7360/xmm7360-pci #2: Thinkpad P53 support

hiddenman, 2020-01-07:
Dear @abrasive 

Thank you for your work. I'm still looking for a working solution for my new P53. Here https://github.com/xmm7360/xmm7360-usb-modeswitch/issues/2#issuecomment-566173118  i asked whether anybody had found any working solution.  Still no succes.

I tried this your new  xmm7360-pci and it hanged my P53 :-)

So if you need any testing or other help, just tell me.

Thank you.

abrasive, 2020-01-08:
Thanks for testing! Can you test it on a VT with logging turned up and see what the crash message is?

- Ctrl-Alt-F1 and login
- `sudo dmesg -n 7`
- load the module

hiddenman, 2020-01-16:
> Thanks for testing! Can you test it on a VT with logging turned up and see what the crash message is?
> 
>     * Ctrl-Alt-F1 and login
> 
>     * `sudo dmesg -n 7`
> 
>     * load the module

Sorry for the delay. Actually, there was a huge dump inside dmesg, then i suspended my notebook, resumed and it didn't respond after that.

Now i just updated to the HEAD, loaded and got his error:

[13832.122530] xmm7360 0000:53:00.0: unknown modem status: 0xffffffff
[13832.122640] xmm7360: probe of 0000:53:00.0 failed with error -22

abrasive, 2020-01-16:
The status 0xffffffff means the modem is not on the PCI bus. Have you been 
using the xmm2usb script or juhovh's kernel module?

hiddenman, 2020-01-16:
> The status 0xffffffff means the modem is not on the PCI bus. Have you been using the xmm2usb script or juhovh's kernel module?

Yes, i tried both of them several times. They said the modem had been reset, but it didn't uppear as USB device. What can i do with that? How can i return it to the PCI bus?

hiddenman, 2020-01-16:
> Easiest is to reboot
> […](#)

Thank you. I even removed the usb module. That helped. Now it says:

[    5.158292] xmm7360 0000:53:00.0: modem is ready

Anything else i can test right now?  Can i access the modem somehow (mmcli can't see it) or the module is not ready yet for this?

Anyway, thank you for your work!

hiddenman, 2020-01-16:
Aah, sorry, i should follow the documentation :-)

hiddenman, 2020-01-16:
Great news. It seems to be working!  I see the wwan0 interface and a lot of data in the terminal. So i'm going to insert a working SIM card (current is blocked) and test it again.

That's a great news for the Lenovo owners!

Thank you very much!

hiddenman, 2020-01-16:
Btw, are you going to integrate this with ModemManager somehow? So it will work seamlessly everywhere.

abrasive, 2020-01-17:
Yes, my intention has been to provide the fundamental information to get the modem working. I'm not the right person to do ModemManager integration (and I don't have time, either), so #4 is open for the time being.

There is also a suggestion that Intel might finally publish their Linux drivers now. My understanding is that they wrote them but didn't release them for some reason. That'd be my ideal outcome!

nlincke, 2020-02-24:
Any news regarding the linux driver?

nlincke, 2020-02-24:
I will open a new thread...

neurotensin, 2020-03-28:
I can confirm this works on Lenovo p53 thinkpad.

Need to start with no defaultroute (-n) and set gateway manually. 

Using apn for Google Fi.

Once slight niggle it seems to mess up the AX200 wifi - but that might be user error.

Thanks @abrasive for getting it working!!

masterdipesh, 2020-06-15:
Hi @abrasive 
I have Lenovo p53 thinkpad and it is not working.

**Below is the lspci -nnvk output**

53:00.0 Non-VGA unclassified device [0000]: Intel Corporation XMM7360 LTE Advanced Modem [8086:7360] (rev 01)
	Flags: fast devsel, IRQ 18
	Memory at ae300000 (64-bit, non-prefetchable) [size=4K]
	Memory at ae301000 (64-bit, non-prefetchable) [size=1K]
	Capabilities: <access denied>
	Kernel modules: xmm7360

**Below is the message i get when i do dmesg | grep xmm**

[    6.528428] xmm7360 0000:53:00.0: enabling device (0000 -> 0002)
[    6.530895] xmm7360 0000:53:00.0: unknown modem status: 0xbadc0ded
[    6.533989] xmm7360: probe of 0000:53:00.0 failed with error -22

If you can help on this that will be great

unituebingen, 2020-06-18:
As a result of my usage on a P53, I feel that it makes a difference when dual booting, weather a WIN10 had the modem activated or disabled when it is shut down. When I shut down WIN10 with WWAN deactivated, it fails to raise the interface on Ubuntu 2004. Rebooting into Windows and changing the modems state seems to make it somehow available again. Please let me know if I can contribute anything regarding this behaviour.

HSPDev, 2020-10-21:
**Confirmed working on new P53 with Fibocom-L850-GL**

I used

```
sudo apt install build-essential python3-pyroute2 python3-configargparse git
git clone https://github.com/xmm7360/xmm7360-pci.git
cd xmm7360-pci
make && make load
sudo python3 rpc/open_xdatachannel.py --apn www.internet.mtelia.dk
sudo ip link set wwan0 up
```

Of course, UEFI boot had to be disabled in BIOS. I usually turn it off and unload the kernel module, when I'm not using the radio using 
```
sudo ip link set wwan0 down
sudo rmmod xmm7360 #from same path as loaded
```

Check if its loaded:
```
lsmod | grep xmm7360
```

Anyway, it works fine for me. I feel like I should be getting faster speeds given I'm in Denmark and 4G LTE is widespread, but I'm unsure if it's the driver bottlenecking somehow or just the P53 / The business subscription I use.

I'm gonna try a portable hotspot to compare. I have a new Netgear Nighthawk inbound for that.

neurotensin, 2020-10-22:
It previously worked (through April)_ but now I am getting:
xmm7360 0000:53:00.0: modem is ready
xmm7360 0000:53:00.0: Could not bring up command ring
xmm7360: probe of 0000:53:00.0 failed with error -110

I am unaware of changes, but of course with corona not been anywhere since Mar!!! 

Any ideas on how to debug?

tgxn, 2021-03-09:
Modem Manager Support has an open PR that I want to look at. 🎉 #66 

@neurotensin @dipesh-p If you're still having issues could you please open a new issue with appropriate details! Thanks :)

closed 2021-03-09 by tgxn