xmm7360/xmm7360-pci #5: Support for HP zbook 14u

strife242, 2020-01-08:
Hello,

First I as the others also would like to thank for the awesome work. I just decided to check into the mobile broadband support again today, and found this.

It loads fine, when trying to run the "open_xdatachannel.py" however I get the following trace.
The interface is created and I get a link-local IPv6 address but nothing else. This may be related to the SIM card as I don't know the status of it so I will try again with a new card tomorrow. Meanwhile, if the SIM is not the issue maybe you find the trace useful anyway.

b'1600000002040000001602040000003011000100020400000000'
b'1600000002040000001602040000002511000100020400000000'
b'1600000002040000001602040000005311000100020400000000'
b'1600000002040000001602040000002411000100020400000000'
unsolicited: b'020100020400000002020400000001020400000000'
b'1600000002040000001602040000003a11000100020400000000'
b'1600000002040000001602040000002611000100020400000000'
b'1600000002040000001602040000000111000100020400000000'
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
FCC lock: state 1 mode 0
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000020200000204000000000204000000000204000000000204000000000204000000010204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000010204000000000204000000000204000004040204000000010204000000000204000000010204000000000204000000005514020400000 [... cut ...] 20400000000020400000000020400000007020400000001020400000002020400000002020400000001020400000002020400000000'
unsolicited: b'0201000204000000ff020400000001020400000000'
b'2500000002040000002502040000004711000101020411000101020100020400000000020400000000'
(IPv4Address('0.0.0.0'), IPv4Address('0.0.0.0'), IPv4Address('0.0.0.0'))
{'v4': [], 'v6': []}
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 47, in <module>
    address=ip_addr)
NameError: name 'ip_addr' is not defined

strife242, 2020-01-09:
Hello again. Just wanted to confirm, with a fresh SIM card and active data plan this works (almost) like a charm. The error message in the original post can happen still, but after running open_xdatachannel.py a few more times it succeeds in taking the interface up and retrieving an IP-address.

Specs on computer is:
HP ZBook 14u G5
Kubuntu 18.10, Linux version 4.18.0-25-generic

One note though, you may have to add DNS-servers manually as they seem not to be retrieved automatically.

Once again, awesome work!

abrasive, 2020-01-10:
This race is the same as #7 and #9, which is probably executing commands before the modem is ready.
We should be waiting for certain unsolicited events before proceeding with certain commands, but I don't know which are the important ones. I can't reproduce this problem here, so can somebody with the issue perhaps figure it out?

I have dug the unsolicited event names out of the firmware and updated the rpc code to print everything nicelier. You should now be able to see which functions are failing (they will return codes like 0xffffff78 instead of 0x0) as well as the meaning and timing of the indications.

If someone wants to just throw down a log of it failing, and a log of it succeeding, then we can probably work it out. Note that this might contain personal information? I don't see an IMEI or anything but I can't be sure.

e-aranda, 2020-01-10:
I attach the result of python3 /rpc/open_xdatachannel.py
[result.log](https://github.com/xmm7360/xmm7360-pci/files/4046388/result.log)

abrasive, 2020-01-11:
Thanks. Pushed a possible fix, does it work for you?
This now waits until it's allowed to do the attach before proceeding to do so.

e-aranda, 2020-01-11:
Sorry I have not understood your last sentence. 

First time executing /rpc/open_xdatachannel.py there is no error, but it "hangs"? After waiting a while I have interrupted. The result was:
[newresul.log](https://github.com/xmm7360/xmm7360-pci/files/4049148/newresul.log)

Then, I have tried again and now is saying

    Attach failed - waiting to see if we just weren't ready

e-aranda, 2020-01-11:
The interface wwan0 has not been created, and while /rpc/open_xdatachannel.py is running I tried another instance of /rpc/open_xdatachannel.py, I get

    Traceback (most recent call last):
      File "rpc/open_xdatachannel.py", line 9, in <module>
        r = rpc.XMMRPC()
      File "/home/earanda/xmm7360-pci/rpc/rpc.py", line 18, in __init__
        self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
    OSError: [Errno 16] Device or resource busy: '/dev/xmm0/rpc'

abrasive, 2020-01-11:
@e-aranda Does your SIM need a PIN? You can try using `AT+CPIN` via `/dev/ttyXMM1` if it does.

e-aranda, 2020-01-11:
I hace disabled PIN in my SIM. I think my APN needs a user and password, however I don't need to introduce un Windows, nor the APM.

abrasive, 2020-01-11:
@e-aranda You definitely need to set the APN at least here, this driver doesn't know how to ask the SIM for the default details to use. If I read the log right, you're with Movistar? So in `open_xdatachannel.py`, replace `telstra.internet` with `telefonica.es` or whatever, and try again.

e-aranda, 2020-01-11:
Yes, I replace `telstra.internet` with `movistar.es` but the result is the same. In this page 

https://www.dell.com/support/article/es/es/esbsdt1/sln60881/3g-mobile-broadband-modem-how-to-configure-and-use-3g-on-a-dell-pc-using-the-ubuntu-operating-system?lang=en

there is a list of internet providers and it says that `telstra` does not need user and password but need to dial `*99#`, while `movistar.es` does not need dialing but use user and password.

I am sorry, I am completely new about these things and don't know what to try, but if you need to check anything, just say it.

woopstar, 2020-03-02:
Fixed by #26

massimoruoroch, 2021-01-25:
Fix works on Lenovo T15

tgxn, 2021-03-10:
#26 has been merged, hopeful; for fix, otherwise please open new issue :)

closed 2021-03-10 by tgxn