xmm7360/xmm7360-pci #6: Error loading module on X1 Carbon 6th Gen

trthomps, 2020-01-08:
Compiled per README, dmesg shows this error on module load:
```
[  273.450257] xmm7360 0000:03:00.0: enabling device (0000 -> 0002)
[  273.450791] xmm7360 0000:03:00.0: unknown modem status: 0xffffffff
[  273.451244] xmm7360: probe of 0000:03:00.0 failed with error -22
```

Kernel:
```
Linux travis-x1-carbon 5.4.8-1-ck-skylake #1 SMP PREEMPT Sun, 05 Jan 2020 11:27:56 +0000 x86_64 GNU/Linux
```

OS: Arch
Commit: 022c999

abrasive, 2020-01-08:
Did you happen to load the USB mode switcher since rebooting? What is the output of `lspci -vs 3:0`?

trthomps, 2020-01-09:
Yes I did, good catch.  I will try rebooting now.

Here's the output ```
03:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
	Subsystem: Intel Corporation XMM7360 LTE Advanced Modem
	Flags: fast devsel, IRQ 18
	[virtual] Memory at ee200000 (64-bit, non-prefetchable) [size=4K]
	[virtual] Memory at ee201000 (64-bit, non-prefetchable) [size=1K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=4/4 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Latency Tolerance Reporting
	Capabilities: [150] L1 PM Substates
	Kernel modules: xmm7360_usb
```

trthomps, 2020-01-09:
OK, that worked, but it kernel panic'ed and locked up my machine. So I rebooted again and loaded the module from the console, and even at 4k I missed over 50% of the output.  I don't have Kdump setup on this machine so it'll probably take me a bit to get the log for you :sweat_smile:

abrasive, 2020-01-15:
There've been a few crash fixes in the module, might be worth trying again if you have time

fredr, 2020-01-26:
I got it running on my X1 carbon 6th gen today. Running Debian 10

chemeris, 2020-01-26:
Loads fine on my ThinkPad X1 Carbon 6th (20KH006JRT) on Ubuntu 18.04.03 64-bit, kernel 5.3.0-26-generic with today's version (0060149958d00b9cec87b73cd610c136f69e5275).

DNS management needs better integration with `systemd` but that's another topic...

bitcoin-software, 2020-02-24:
6th Gen Thinkpad on Fedora, get this: FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'

trthomps, 2020-03-11:
Sorry for not responding for so long, this works on my X1 now as well.  Thanks!

closed 2020-03-11 by trthomps