xmm7360/xmm7360-pci #7: Many thanks got this working on Ubuntu 19.10!

techjump, 2020-01-09:
Hi,

I managed to get this working on my HP Elite X2 where all previous attempts using the usb mode switcher had failed.

Happy to do some testing if need be.

As a note to others I had to manually setup DNS nameservers to physically use the connection other than pings etc.

e-aranda, 2020-01-09:
Hi, I also have a HP Elite X2 G4 with Ubuntu 19.10 but `make load` does not work. Compilation with `make` seems Ok, 

    make -C /lib/modules/5.3.0-26-generic/build M=/opt/software/drivers/xmm7360-pci modules
    make[1]: se entra en el directorio '/usr/src/linux-headers-5.3.0-26-generic'
      CC [M]  /opt/software/drivers/xmm7360-pci/xmm7360.o
      Building modules, stage 2.
      MODPOST 1 modules
      CC      /opt/software/drivers/xmm7360-pci/xmm7360.mod.o
      LD [M]  /opt/software/drivers/xmm7360-pci/xmm7360.ko
    make[1]: se sale del directorio '/usr/src/linux-headers-5.3.0-26-generic'

but when `make load` gives:

    sudo /sbin/rmmod xmm7360
    rmmod: ERROR: Module xmm7360 is not currently loaded
    make: [Makefile:18: load] Error 1 (no tiene efecto)
    sudo /sbin/insmod xmm7360.ko
    insmod: ERROR: could not insert module xmm7360.ko: Operation not permitted
    make: *** [Makefile:19: load] Error 1

What is your kernel version?

techjump, 2020-01-09:
Exactly the same kernel version. 5.3.0-26
You don't have secure boot enabled do you?

e-aranda, 2020-01-10:
Yes, I have.

e-aranda, 2020-01-10:
Ok, I have disabled secure boot and now module is loaded. However, I get the same problem as #5 and  #9, but after trying many times I don't get a IP address

abrasive, 2020-01-10:
Thanks for the report! Should probably add a thing that does the nameservers hey. But there's no way to do it that isn't rude...

closed 2020-01-15 by abrasive