xmm7360/xmm7360-pci #8: Kernel Panic on Elitebook x360

J0n4t4n, 2020-01-09:
So after successfully loading the module the first time and running into the previous issue #3 , I wanted to now try again.
This time though the module was not able to load, but instead I had my Laptop freeze up on trying to insert the module.
I grabbed a log like you described in #2, pictures attached.

[image]
[image]

abrasive, 2020-01-10:
Can you do an `objdump -S xmm7360.ko` (assuming it's still the same compile that crashed) and work out which code is at `xmm7360_irq0+0x51`? Or just paste the disassembly somewhere

J0n4t4n, 2020-01-10:
I wanted to do that but I didn't have the failing compile anymore.
Did a new one and it worked this time (Still checked the objdump and there was no xmm7360_irq0+0x51 anywhere).

After the module loaded successfully this time, I ran the python script multiple times.
The first time I had the same problem as #9, the second time the script just ran indefinetly and after the 5th run I finally had an IP and it is now working perfectly (after manually setting a DNS server).

closed 2020-01-10 by J0n4t4n