xmm7360/xmm7360-pci #9: name 'ip_addr' is not defined in first script call

hinaloe, 2020-01-09:
I using Arch Linux on ThinkPad X395 (with L850-GL).

in Arch Official basic kernel (`Linux version 5.4.8-arch1-1 (linux@archlinux) (gcc version 9.2.0 (GCC)) #1 SMP PREEMPT Sat, 04 Jan 2020 23:46:18 +0000`), `open_xdatachannel.py` throws the NameError (like #5 ). but, re-call the script, that success and connected internet.
I can't determine if it's the same problem.

### outputs

1st time
```
 ~/w/xmm7360-pci (master)> sudo python3 ./rpc/open_xdatachannel.py
b'1600000002040000001602040000003011000100020400000000'
b'1600000002040000001602040000002511000100020400000000'
b'1600000002040000001602040000005311000100020400000000'
b'1600000002040000001602040000002411000100020400000000'
b'1600000002040000001602040000003a11000100020400000000'
b'1600000002040000001602040000002611000100020400000000'
b'1600000002040000001602040000000111000100020400000000'
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
FCC lock: state 0 mode 2
b'1c00000002040000001c02040000019011000101020411000101020400000000'
b'1c00000002040000001c0204000001921100010102041100010102046091f1df'
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000102040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000102040000000002040000000002040000040402040000000102040000000002040000000102040000000002040000000055140204 [... cut ...]
b'b80000000204000000b802040000007d110001000201000204000000320204000000010204000000030204000000050201000201fa0201000201fa0201000201000201ff0201ff0201000201120201060201080201fe0201fe02019602010302010202010402010102011f02010902010302010202010402010502040000000155140204000000160204000000020ae52f9b000000000000000000000000000000000000020400000000020400000000020400020017020400000000'
```

-------

In addition, I usually use a [patched kernel](https://aur.archlinux.org/packages/linux-amd/), and the system freezes after a kernel message like the image below.

[image]

abrasive, 2020-01-10:
The script should probably be waiting for certain unsolicited events eg. that the radio is up, attached to network, etc. Unfortunately I have no idea what any of those messages are!

abrasive, 2020-01-10:
"Scheduling while atomic" should be fixed, thanks for that report

abrasive, 2020-01-10:
Closing as the remaining issue falls under #5

closed 2020-01-10 by abrasive