xmm7360/xmm7360-pci #10: Issues with Thinkpad X1 Yoga 4th Gen

Gottox, 2020-01-13:
Thanks for your awesome work! Unfortunately I was unable to get the modem to work on my Thinkpad X1 Yoga 4th Gen.

## dmesg
```
[   77.671574] xmm7360: loading out-of-tree module taints kernel.
[   77.672659] xmm7360 0000:02:00.0: enabling device (0000 -> 0002)
[   77.673618] xmm7360 0000:02:00.0: modem is ready
```
## Console Output:
```
~/src/xmm7360-pci(master ✗) sudo python3 rpc/open_xdatachannel.py
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x58b728e7
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c02040000019211000101020411000101020441701c8f'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
 # disable aeroplane mode if had been FCC-locked. first and second args are probably don't-cares
 r.execute('UtaModeSetReq', rpc.pack('LLL', 0, 15, 1))
 
-r.execute('UtaMsCallPsAttachApnConfigReq', rpc.pack_UtaMsCallPsAttachApnConfigReq("telstra.internet"), is_async=True)
+r.execute('UtaMsCallPsAttachApnConfigReq', rpc.pack_UtaMsCallPsAttachApnConfigReq("web.vodafone.de"), is_async=True)
 
 attach = r.execute('UtaMsNetAttachReq', rpc.pack_UtaMsNetAttachReq(), is_async=True)
 _, status = rpc.unpack('nn', attach['body'])
```

abrasive, 2020-01-13:
Thanks for testing! Time for some sketchy debugging of ordering.

Does it work if you try running `open_xdatachannel.py` multiple times?

Does adding a delay help - say 20 seconds after UtaModeSetReq?

Gottox, 2020-01-13:
> Does it work if you try running `open_xdatachannel.py` multiple times?
No change:
> Does adding a delay help - say 20 seconds after UtaModeSetReq?
Nope.

I had some issues with the usb-mode switch too, Could it be an hardware issue? Also how do I find out the firmware version?

abrasive, 2020-01-14:
I've added some more stuff:

- The `UtaModeSet` request now waits until it's actually completed
- Firmware version is printed if running `rpc/rpc.py`

My firmware version is `18500.5001.11.02.22.93`. I have booted the modem with Windows, which once it detected my carrier, flashed some supplemental NVRAM variables. But yours connected in USB mode, no?

After running the init code in `rpc/rpc.py`, you could also try bringing up a PPP link - see the `test.sh` and `test.chat` provided up until 95e89c6

Gottox, 2020-01-14:
```
Firmware version: bytearray(b'18500.5001.00.01.20.86')
```
Unfortunately, I don't have windows on my notebook anymore. Is there a way to update it with linux?

m7mdcc, 2020-01-14:
@Gottox  , There is tool that work for me on windows , they have linux version also .. 
https://github.com/projectceladon/tools
you need to pass firmware files to Downloadtool executable file included with this package 
see this : https://whrl.pl/RfXQlT

Gottox, 2020-01-14:
@m7mdcc cool! I'm able to extract the *.flz/*.fls files from the [Lenovo Download Page](https://support.lenovo.com/de/en/downloads/DS542005) But I'm not sure how to install the using the tool.

abrasive, 2020-01-14:
When you reset the modem (eg via ACPI) it goes into bootloader mode for 5 
sec and shows up as a USB device. 8086:07f5 if I recall correctly

Gottox, 2020-01-14:
`Here be dragons.`

Gottox, 2020-01-14:
```
libusb: error [_get_usbfs_fd] File doesn't exist, wait 10 ms and try again
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/019: No such file or directory
Error> [1578996810.992] libusb_open_wrapper:983: Error opening device (001/019) -4: Device disconnected

Error> [1578996810.992] GenericUSB_open:928:  Could not open device 1 :'(

Error> [1578996810.992] GenericUSB_open:934:  Error 2: No such file or directory

Error> Communication error 2. Line 937
Segmentation fault
```

m7mdcc, 2020-01-14:
Do you have 860 or 850 ? 

for 850 go to  : https://support.lenovo.com/us/en/downloads/DS503062 ..

try with only :  ../bin/downloadTool /tool/pathtoflsfile

Gottox, 2020-01-14:
I'm retrying that later with fedora instead of debian

Gottox, 2020-01-14:
@m7mdcc same with 850. I had no time to boot to fedora yet.

I modified the usb modeswitch script to only reset the modem:

```diff
diff --git a/xmm2usb b/xmm2usb
index 67a75ad..5327d77 100755
--- a/xmm2usb
+++ b/xmm2usb
@@ -30,15 +30,14 @@ for path in `find /sys/devices/pci0000:00 -name vendor`; do
        modprobe acpi_call || \
                die "Could not load acpi_call module. Is it missing?"
 
-       printf "Disabling PCIe link...\n" >&2
-       setpci -s "${bridge_id}" "CAP_EXP+10.w=0052" || \
-               die "Could not run setpci"
+       #printf "Disabling PCIe link...\n" >&2
+       #setpci -s "${bridge_id}" "CAP_EXP+10.w=0052" || \
+       #       die "Could not run setpci"
 
        printf "%s" "${acpi_path}._RST" > /proc/acpi/call || \
                die "Could not reset PCI"
 
        echo "OK!"
-
        exit 0
 done
```

Gottox, 2020-01-15:
I'm not able to update the firmware atm. I think I need a windows box for that. will try that at the end of the week.

m7mdcc, 2020-01-15:
@Gottox , Sorry if that doesn't help  you in your current build . but can you look here may its help ? 
https://www.quectel.com/UploadImage/Downlad/Quectel_WCDMA_UGxx_DownloadTool_Linux_User_Guide_V1.2.pdf

abrasive, 2020-01-15:
@Gottox, did you succeed in bringing it up in USB mode with the same SIM card?

Gottox, 2020-01-15:
Yes, but actually it took many reboots to do so.

And since the update to NetworkManager 1.22 I was unable to connect, but that looks like a different issue to me.

Gottox, 2020-01-15:
To answer @m7mdcc's question, it looks like the 860 one, at least the compatibility list on the lenovo page states the support for my hardware.

Gottox, 2020-01-15:
Just to make sure: The firmware update can be done plainly through USB? - I have a windows PC and [an m.2 to usb connector](https://www.amazon.de/gp/product/B07DC35G1X/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1) that works with this particular card.

abrasive, 2020-01-15:
Hmm, yes, but it's not clear whether you can do that with that setup...

There is an MBIM command you can send to the modem via USB that will
kick it into bootloader mode. ModemManager can send it. The Windows
driver sends the equivalent RPC calls over PCI, then does the update via
USB. But the Windows driver doesn't support the main modem in USB mode,
as far as I know... so the updater /might/ insist on seeing it over PCI
first. Or it might just accept it if it only sees it in bootloader mode.

Gottox, 2020-01-18:
Okay, I'm currently back in Ubuntu. Still unable to update the firmware. Currently downloadTool hangs:
[image]

m7mdcc, 2020-01-23:
@Gottox did you try that on windows setup ?

m7mdcc, 2020-01-23:
> particular

I'm doing this like that .

Gottox, 2020-01-23:
I tried it using a windows PE setup, this didn't work. I think I need a full installation.

Currently I don't have time to have the TP offline for half a day to set up windows. I think I'll do that at the beginning of February. Then I'll come back to this issue.

Gottox, 2020-02-15:
Doing an upgrade under windows now. Boy, is windows [awful](https://twitter.com/Gottox/status/1228751673679634432)...

Gottox, 2020-02-15:
Okay, firmware is updated, but the problem persists:

```
 sudo python3 rpc/rpc.py | grep Firmware
Firmware version: bytearray(b'18500.5001.03.01.20.85')
```

Gottox, 2020-02-16:
PEBCAK...

I...

I didn't disable the PIN on the card. :disappointed:

closed 2020-02-16 by Gottox