xmm7360/xmm7360-pci #11: insmod "killed" on Thinkpad X1 Yoga 7th gen

dabrowskiw, 2020-01-14:
Hello,

first of all thanks a lot for the awesome work, I've been waiting for this since I got my X1Y!

Unfortunately, when I try to insmod xmm7360.ko, I just get "Killed" as response. I then cannot rmmod xmm7360, since "rmmod: ERROR: Module xmm7360 is in use". lsmod tells me: "xmm7360                28672  1", which is not very informative.

dmesg tells me that something went wrong while loading the module: 

[  451.303385] xmm7360: loading out-of-tree module taints kernel.
[  451.303433] xmm7360: module verification failed: signature and/or required key missing - tainting kernel
[  451.303992] xmm7360 0000:02:00.0: enabling device (0000 -> 0002)
[  451.304301] xmm7360 0000:02:00.0: pci_alloc_irq_vectors
[  451.304304] ------------[ cut here ]------------
[  451.304310] WARNING: CPU: 0 PID: 4529 at kernel/workqueue.c:2911 __flush_work.cold.53+0x1f/0x29
[  451.304310] Modules linked in: xmm7360(OE+) fuse bnep intel_rapl x86_pkg_temp_thermal intel_powerclamp nls_ascii coretemp snd_hda_codec_hdmi nls_cp437 vfat fat kvm snd_usb_audio irqbypass cdc_ether snd_usbmidi_lib usbnet snd_rawmidi crct10dif_pclmul snd_seq_device mii snd_soc_skl crc32_pclmul snd_soc_skl_ipc ghash_clmulni_intel snd_soc_sst_ipc intel_cstate snd_soc_sst_dsp joydev snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi intel_uncore snd_soc_core snd_hda_codec_realtek intel_rapl_perf snd_compress snd_hda_codec_generic efi_pstore wmi_bmof hid_multitouch serio_raw snd_hda_intel btusb btrtl tpm_crb btbcm snd_hda_codec btintel iTCO_wdt bluetooth pcspkr efivars iTCO_vendor_support iwlwifi uvcvideo snd_hda_core videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_hwdep snd_pcm
[  451.304329]  videodev drbg cfg80211 ansi_cprng snd_timer media ecdh_generic ucsi_acpi typec_ucsi tpm_tis processor_thermal_device idma64 tpm_tis_core typec intel_soc_dts_iosf intel_pch_thermal mei_me tpm pcc_cpufreq mei rng_core thinkpad_acpi nvram snd soundcore ac rfkill battery int3403_thermal int340x_thermal_zone acpi_pad acpi_tad evdev int3400_thermal acpi_thermal_rel efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 fscrypto ecb btrfs xor zstd_decompress zstd_compress xxhash raid6_pq libcrc32c crc32c_generic hid_microsoft hid_sensor_custom hid_sensor_hub intel_ishtp_hid wacom usbhid hid_generic i915 crc32c_intel i2c_algo_bit drm_kms_helper xhci_pci aesni_intel xhci_hcd psmouse e1000e drm aes_x86_64 nvme crypto_simd usbcore cryptd glue_helper thunderbolt i2c_i801 nvme_core i2c_hid intel_ish_ipc
[  451.304351]  hid intel_lpss_pci intel_ishtp intel_lpss usb_common thermal wmi video button
[  451.304355] CPU: 0 PID: 4529 Comm: insmod Tainted: G           OE     4.19.0-6-amd64 #1 Debian 4.19.67-2+deb10u2
[  451.304356] Hardware name: LENOVO 20QGS01M00/20QGS01M00, BIOS N2HET37W (1.20 ) 09/11/2019
[  451.304357] RIP: 0010:__flush_work.cold.53+0x1f/0x29
[  451.304358] Code: e9 27 04 00 0f 0b e9 7a d3 ff ff 48 c7 c7 78 0c 84 90 e8 d6 27 04 00 0f 0b e9 71 d6 ff ff 48 c7 c7 78 0c 84 90 e8 c3 27 04 00 <0f> 0b 45 31 ed e9 5b d6 ff ff 49 8d b4 24 b0 00 00 00 48 c7 c7 28
[  451.304359] RSP: 0018:ffffb2698536fa08 EFLAGS: 00010246
[  451.304359] RAX: 0000000000000024 RBX: ffff8e46a4a80048 RCX: 0000000000000006
[  451.304360] RDX: 0000000000000000 RSI: 0000000000000096 RDI: ffff8e46ac2166b0
[  451.304360] RBP: ffff8e46a4a80048 R08: 0000000000000505 R09: 0000000000000004
[  451.304361] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000000
[  451.304361] R13: 0000000000000001 R14: ffffffff8fa98420 R15: 0000000000000000
[  451.304362] FS:  00007f80d5a12480(0000) GS:ffff8e46ac200000(0000) knlGS:0000000000000000
[  451.304362] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  451.304362] CR2: 000055951ddd71a8 CR3: 00000004341bc006 CR4: 00000000003606f0
[  451.304363] Call Trace:
[  451.304367]  ? xmm7360_net_xmit+0x190/0x190 [xmm7360]
[  451.304368]  ? get_work_pool+0x40/0x40
[  451.304369]  __cancel_work_timer+0x10a/0x190
[  451.304372]  ? _dev_err+0x6c/0x90
[  451.304373]  ? xmm7360_net_xmit+0x190/0x190 [xmm7360]
[  451.304374]  xmm7360_dev_deinit+0x23/0x170 [xmm7360]
[  451.304376]  xmm7360_probe+0x234/0x340 [xmm7360]
[  451.304378]  local_pci_probe+0x41/0x90
[  451.304379]  pci_device_probe+0xf5/0x1b0
[  451.304380]  really_probe+0x23e/0x390
[  451.304382]  driver_probe_device+0xb3/0xf0
[  451.304383]  __driver_attach+0xea/0x110
[  451.304384]  ? driver_probe_device+0xf0/0xf0
[  451.304385]  bus_for_each_dev+0x77/0xc0
[  451.304387]  ? klist_add_tail+0x3b/0x70
[  451.304388]  bus_add_driver+0x152/0x230
[  451.304390]  ? xmm7360_probe+0x340/0x340 [xmm7360]
[  451.304391]  driver_register+0x6b/0xb0
[  451.304392]  ? xmm7360_probe+0x340/0x340 [xmm7360]
[  451.304394]  do_one_initcall+0x46/0x1c3
[  451.304396]  ? free_unref_page_commit+0x91/0x100
[  451.304397]  ? _cond_resched+0x15/0x30
[  451.304398]  ? kmem_cache_alloc_trace+0x155/0x1d0
[  451.304400]  do_init_module+0x5a/0x210
[  451.304401]  load_module+0x2118/0x2390
[  451.304403]  ? __do_sys_finit_module+0xad/0x110
[  451.304404]  __do_sys_finit_module+0xad/0x110
[  451.304406]  do_syscall_64+0x53/0x110
[  451.304407]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  451.304408] RIP: 0033:0x7f80d5b2cf59
[  451.304409] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 07 6f 0c 00 f7 d8 64 89 01 48
[  451.304409] RSP: 002b:00007ffeb6ddaa48 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  451.304410] RAX: ffffffffffffffda RBX: 000055951ddd47a0 RCX: 00007f80d5b2cf59
[  451.304410] RDX: 0000000000000000 RSI: 000055951cf8b3f0 RDI: 0000000000000003
[  451.304411] RBP: 000055951cf8b3f0 R08: 0000000000000000 R09: 00007f80d5bf7000
[  451.304411] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[  451.304411] R13: 000055951ddd4760 R14: 0000000000000000 R15: 0000000000000000
[  451.304412] ---[ end trace c88992ef0c24071c ]---
[  451.304418] BUG: unable to handle kernel paging request at 0000000000022b08
[... 30 lines cut ...]
[  451.304449]  do_one_initcall+0x46/0x1c3
[  451.304450]  ? free_unref_page_commit+0x91/0x100
[  451.304451]  ? _cond_resched+0x15/0x30
[  451.304452]  ? kmem_cache_alloc_trace+0x155/0x1d0
[  451.304453]  do_init_module+0x5a/0x210
[  451.304454]  load_module+0x2118/0x2390
[  451.304456]  ? __do_sys_finit_module+0xad/0x110
[  451.304457]  __do_sys_finit_module+0xad/0x110
[  451.304458]  do_syscall_64+0x53/0x110
[  451.304460]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  451.304460] RIP: 0033:0x7f80d5b2cf59
[  451.304461] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 07 6f 0c 00 f7 d8 64 89 01 48
[  451.304462] RSP: 002b:00007ffeb6ddaa48 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  451.304463] RAX: ffffffffffffffda RBX: 000055951ddd47a0 RCX: 00007f80d5b2cf59
[  451.304463] RDX: 0000000000000000 RSI: 000055951cf8b3f0 RDI: 0000000000000003
[  451.304464] RBP: 000055951cf8b3f0 R08: 0000000000000000 R09: 00007f80d5bf7000
[  451.304464] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[  451.304465] R13: 000055951ddd4760 R14: 0000000000000000 R15: 0000000000000000
[  451.304466] Modules linked in: xmm7360(OE+) fuse bnep intel_rapl x86_pkg_temp_thermal intel_powerclamp nls_ascii coretemp snd_hda_codec_hdmi nls_cp437 vfat fat kvm snd_usb_audio irqbypass cdc_ether snd_usbmidi_lib usbnet snd_rawmidi crct10dif_pclmul snd_seq_device mii snd_soc_skl crc32_pclmul snd_soc_skl_ipc ghash_clmulni_intel snd_soc_sst_ipc intel_cstate snd_soc_sst_dsp joydev snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi intel_uncore snd_soc_core snd_hda_codec_realtek intel_rapl_perf snd_compress snd_hda_codec_generic efi_pstore wmi_bmof hid_multitouch serio_raw snd_hda_intel btusb btrtl tpm_crb btbcm snd_hda_codec btintel iTCO_wdt bluetooth pcspkr efivars iTCO_vendor_support iwlwifi uvcvideo snd_hda_core videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_hwdep snd_pcm
[  451.304478]  videodev drbg cfg80211 ansi_cprng snd_timer media ecdh_generic ucsi_acpi typec_ucsi tpm_tis processor_thermal_device idma64 tpm_tis_core typec intel_soc_dts_iosf intel_pch_thermal mei_me tpm pcc_cpufreq mei rng_core thinkpad_acpi nvram snd soundcore ac rfkill battery int3403_thermal int340x_thermal_zone acpi_pad acpi_tad evdev int3400_thermal acpi_thermal_rel efivarfs ip_tables [... cut ...]  hid_generic i915 crc32c_intel i2c_algo_bit drm_kms_helper xhci_pci aesni_intel xhci_hcd psmouse e1000e drm aes_x86_64 nvme crypto_simd usbcore cryptd glue_helper thunderbolt i2c_i801 nvme_core i2c_hid intel_ish_ipc
[  451.304493]  hid intel_lpss_pci intel_ishtp intel_lpss usb_common thermal wmi video button
[  451.304496] CR2: 0000000000022b08
[  451.304497] ---[ end trace c88992ef0c24071d ]---

Unfortunately, this is way beyond my skill - any ideas what might have gone wrong and what I can do?

Gottox, 2020-01-15:
I had similiar issues with ubuntu 16.04. Under VoidLinux with kernel 5.4.11_1 it worked.

abrasive, 2020-01-15:
This looks like two issues:

- The driver isn't getting the number of interrupts it requested. We only use one anyway at the moment, so this should be fixed in 0060149
- The bailout path when this fails causes a crash elsewhere. That will need to be fixed... later.

Can you check if 0060149 fixes this?

dabrowskiw, 2020-01-15:
Thanks, the module loads without any trouble now!

closed 2020-01-15 by dabrowskiw