xmm7360/xmm7360-pci #12: Works with Lenovo ThinkPad X390 under Debian 10 - except automatic DNS

DDvO, 2020-01-14:
Thanks @abrasive for this new solution - very cool!
It works much better for me than https://github.com/xmm7360/xmm7360-usb-modeswitch

Just two issues.

* As already mentioned by another user in #7, 
I need to set up DNS manually.
I do get some direct routing - `ip route`  yields 
```
default dev wwan0 proto static
```
From two of the many lines of output of the Python script (in my case for `web.vodafone.de`):
 ```
(IPv4Address('0.0.0.82'), IPv4Address('31.213.168.1'), IPv4Address('100.98.7.178'))
{'v4': [IPv4Address('139.7.30.125'), IPv4Address('139.7.30.126'), IPv4Address('139.7.30.125')], 'v6': [IPv6Address('2a01:860:0:300::fffe:53'), IPv6Address('2a01:860:0:300::ffff:53')]}
```
I can copy and use one of the DNS server IP addresses. For instance:
```
echo 'nameserver 139.7.30.126' > /etc/resolv.conf
```
but I wonder why this is not done somehow automatically.

* How is one supposed to close the modem connection via device `wwan0`?
When I use
```
sudo ip link set wwan0 down
```
the device disappers form the output of `ifconfig` 
but the next connection attempt fails with
```
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 64, in <module>
    address=ip_addr)
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/iproute/linux.py", line 1448, in addr
    terminate=lambda x: x['header']['type'] ==
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 373, in nlm_request
    return tuple(self._genlm_request(*argv, **kwarg))
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 870, in nlm_request
    callback=callback):
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 376, in get
    return tuple(self._genlm_get(*argv, **kwarg))
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 707, in get
    raise msg['header']['error']
pyroute2.netlink.exceptions.NetlinkError: (17, 'File exists')
```

abrasive, 2020-01-14:
It's not done automatically because managing `resolv.conf` is a Hard Problem eg. putting it back when we're "done". Really this needs to be handled by a daemon like `ModemManager`. I guess we could append them to `resolv.conf` for now?

abrasive, 2020-01-15:
Fixed in 7b12f3f8024e827d4f6b622cb93ba1051625212d (and later made configurable to taste)

closed 2020-01-15 by abrasive

DDvO, 2020-01-15:
I agree that appending to ` /etc/resolv.conf` is at least a viable interim solution.
Just please don't add the same entires again and again (when the connection is closed and later re-opened).
Checking for the existence of `# Added by xmm7360` and not adding entries if present would be a simple possibility, but this would not support changing DNS entries.
A more clever possibility would be: for each DNS entry to be added, check it it is already there and not add it again if so.