xmm7360/xmm7360-pci #13: How to cleanly shut down and reopen the modem connection?

DDvO, 2020-01-14:
How is one supposed to close and re-open the modem connection via device `wwan0`?
When I use
```
sudo ip link set wwan0 down
```
the device disappers form the output of `ifconfig` 
but the next connection attempt fails with
```
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 64, in <module>
    address=ip_addr)
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/iproute/linux.py", line 1448, in addr
    terminate=lambda x: x['header']['type'] ==
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 373, in nlm_request
    return tuple(self._genlm_request(*argv, **kwarg))
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 870, in nlm_request
    callback=callback):
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 376, in get
    return tuple(self._genlm_get(*argv, **kwarg))
  File "/usr/local/lib/python3.7/dist-packages/pyroute2/netlink/nlsocket.py", line 707, in get
    raise msg['header']['error']
pyroute2.netlink.exceptions.NetlinkError: (17, 'File exists')
```

It appears more clean to use `rmmod xmm7360`.

Yet then the Python script gives errors unless one manually re-inserts the kernel module, e.g.,
``` 
insmod xmm7360.ko
```

c3r34lk1ll3r, 2020-01-14:
The problem is that `pyroute2` try to add an already present interface. You can solve the issue adding a `try except` block like this:
```
try:
    ipr.addr('add',index=idx,address=ip_addr)
except Exception:
    print("Already exist")
```
This code is very "crude" but will give you an idea how to solve the issue.

DDvO, 2020-01-14:
Indeed, after replacing in `rpc/open_xdatachannel.py` the statment
```
ipr.addr('add',
        index=idx,
        address=ip_addr)

```
by
```
try:
    ipr.addr('add',
             index=idx,
             address=ip_addr)
except Exception:
    print("Interface already exists")
```

it is sufficient to close the modem interface by `sudo ip link set wwan0 down`.
Then re-opening via the script it works just fine :)

Thanks!

abrasive, 2020-01-15:
Fixed in 706181e by removing IPs (and consequently routes) before beginning.

closed 2020-01-15 by abrasive