xmm7360/xmm7360-pci #14: pc/open_xdatachannel.py fails at first

DDvO, 2020-01-14:
When running `sudo python3 rpc/open_xdatachannel.py` the first time I usually get
```
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x25, 0x0, 0x90, 0x33, 0x1, 0x33, 0x1, 0x0, 0x0
unsolicited: UtaMsSimFullAccessIndCb: 0x0
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 27, in <module>
    attach = r.execute('UtaMsNetAttachReq', rpc.pack_UtaMsNetAttachReq(), is_async=True)
  File "/home/david/sys/software/xmm7360-pci/rpc/rpc.py", line 67, in execute
    resp = self.pump()
  File "/home/david/sys/software/xmm7360-pci/rpc/rpc.py", line 32, in pump
    desc = 'unsolicited: %s' % rpc_unsol_table.xmm7360_unsol[resp['code']]
KeyError: 457
```
Yet it succeeds when simply calling it a second time. 
Somewhat strange, but not a major problem.

abrasive, 2020-01-14:
Inadvertently committed a fix in 2066cdb021102fcc73b6cfb029bec5420864f653 ^_^

Would you mind testing this?

DDvO, 2020-01-15:
Just tested the new version. As before, I tend to get 
```
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x25, 0x0, 0x90, 0x33, 0x1, 0x33, 0x1, 0x0, 0x0
unsolicited: UtaMsSimFullAccessIndCb: 0x0
unsolicited: UtaMsSmsSimMsgCacheFinishedIndCb: 0x0, 0x0
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 27, in <module>
    attach = r.execute('UtaMsNetAttachReq', rpc.pack_UtaMsNetAttachReq(), is_async=True)
  File "/home/david/sys/software/xmm7360-pci/rpc/rpc.py", line 67, in execute
    resp = self.pump()
  File "/home/david/sys/software/xmm7360-pci/rpc/rpc.py", line 32, in pump
    desc = 'unsolicited: %s' % rpc_unsol_table.xmm7360_unsol[resp['code']]
KeyError: 457
```
on the first (not second) connection attempt.

abrasive, 2020-01-15:
@DDvO That line 32 in your backtrace doesn't match 2066cdb?

DDvO, 2020-01-15:
Oops, my `git pull` had not worked as expected.
I've fixed this and now really use the new version.
And it works fine - thanks a lot :+1:

closed 2020-01-15 by DDvO