xmm7360/xmm7360-pci #15: System hangs on resume after suspend due to xmm7360 - workaround exists

DDvO, 2020-01-14:
Whenever the `xmm7360` kernel module is loaded (and maybe some connection is still active) on supspend,
my Debian 10 system hangs on resume with a black screen, and I need to hard-reset it.

Here is a workaround which simply removes the module on suspend:
Add a file `/usr/lib/systemd/system-sleep/re-suspend.sh` containing
```
#!/bin/bash
if [ "$1" == "pre" ]; then
    rmmod xmm7360
fi
```
and use `chmod u+x /usr/lib/systemd/system-sleep/pre-suspend.sh` to make it executable.

abrasive, 2020-01-14:
"Fixed" in 2066cdb021102fcc73b6cfb029bec5420864f653, at least until someone makes suspend/resume handling work properly.

closed 2020-01-14 by abrasive

DDvO, 2020-01-15:
Thanks for your swift reaction also in this topic.
Indeed unloading the module before suspending is not needed anymore for the resume to work.

Yet the module is in an unclean state thereafter: When I call the script I get
```
xmm7360                28672  0
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 30, in <module>
    r = rpc.XMMRPC()
  File "/home/david/sys/software/xmm7360-pci/rpc/rpc.py", line 18, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
OSError: [Errno 16] Device or resource busy: '/dev/xmm0/rpc'
```

A workaround is to manually unload and load it again before calling the script:
````
sudo rmmod xmm7360
sudo insmod xmm7360.ko
````

abrasive, 2020-01-15:
Yes, unfortunately, the "fix" was to remove my not-quite-working
suspend/resume handling ;-) This is tracked as part of #1.