xmm7360/xmm7360-pci #18: Thinkpad X395 - Attach failed again, giving up

ganter799, 2020-01-17:
Hello, I am unable to get my modem working on my Thinkpad X395.

### Console:
```
sudo python3 rpc/open_xdatachannel.py --apn internet
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x2, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x2
FCC lock: state 1 mode 2
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc02040000000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002020000020400000000020400000000020400000000020400000000020400000001020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000001020400000 [... cut ...]
unsolicited: UtaMsNetCellInfoIndCb: 0x0, 0x0, 0x0, 0xffff, 0x0, 0x32, 0x2, 0x0, 0x2, 0x1, 0x1, 0x1, 0x0, 0x0, 0x0, 0x3fe, 0x295d2c4, 0x0, 0x2, 0x106, 0x0
response: 0x0, 0xffffffff, 0x2, 0x2, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x0, 0x2, 0x1, 0x0
ERROR:root:Attach failed again, giving up
```
### dmesg:
```
[  328.049676] xmm7360 0000:04:00.0: enabling device (0000 -> 0002)
[  328.050065] xmm7360 0000:04:00.0: modem is ready
[  341.591523] xmm7360 0000:04:00.0: modem is ready

### rpc.py:
```
response: 0x0, 0x16, bytearray(b'18500.5001.00.01.20.75')
Firmware version: bytearray(b'18500.5001.00.01.20.75')
```

raphaelm, 2020-01-21:
Same here on X1 Carbon Gen7. Noticed that my SIM card has a PIN set and there is nothing in this repository to ask for PINs – could this be the issue?

hinaloe, 2020-01-22:
> Same here on X1 Carbon Gen7. Noticed that my SIM card has a PIN set and there is nothing in this repository to ask for PINs – could this be the issue?

Did you see https://github.com/xmm7360/xmm7360-pci/issues/5#issuecomment-573305213 ?

nmakhotkin, 2020-06-15:
The same error for me on Thinkpad X1 Carbon 6th gen. The SIM doesn't have PIN at all.
I am on Ubuntu 20.04, kernel 5.4.0. syslog says

```kernel: [69434.037057] xmm7360 0000:03:00.0: enabling device (0000 -> 0002)
kernel: [69434.037538] xmm7360 0000:03:00.0: modem is ready
```

Also I tried to send default PIN to SIM (provider gives 0000 on all SIM by default) just in case (`echo "AT+CPIN=\"0000\"" | sudo tee -a /dev/ttyXMM1`) but with no result.

sigint2, 2020-06-25:
Same as @nmakhotkin , no PIN and same error message.

tgxn, 2021-03-10:
Any progress on this @ganter799 @sigint2 or @nmakhotkin  ?

I've merged several PR's that may have resolved these issues.

nmakhotkin, 2021-03-10:
I didn't try it for a long time and don't use the modem now. It was a case for me just last summer :(

tgxn, 2021-03-10:
Hey thanks for the reply mate, I'm going to close off this issue. :) If anyone is still having issues, please create a new issue!

closed 2021-03-10 by tgxn