xmm7360/xmm7360-pci #22: Signal level / access technology status reading

chemeris, 2020-01-26:
Sorry if this is an obvious question, but is there a way to read the signal level and used access technology (3G/4G/etc) from the driver?

c3r34lk1ll3r, 2020-01-28:
In theory, you can access this information using AT interface.

For example,
`echo 'AT+CSQ?' | sudo tee /dev/ttyXMM1`

should return the signal quality.

`echo 'AT+COPS?' | sudo tee /dev/ttyXMM1`

should return the details about the connection ([here](https://m2msupport.net/m2msupport/atcops-plmn-selection/) you can find how to read the value ).

`AT+CLAC` will print all the command supported (although, it seems that not all are working).

chemeris, 2020-01-29:
Thank you, makes sense (=works!).

closed 2020-01-29 by chemeris