xmm7360/xmm7360-pci #23: Working on HP Dragonfly, but little feature request

foehnph, 2020-01-28:
Hi

Thanks for the great development! It is working on my HP Elite Dragonfly 8MK79EA with the XMM7360!
However, I have two specific requests:
1. Could you add a SIM PIN argument?
    I worked around it with the `AT+CLCK=“SC”,1,“YOURPIN”,1` 
2. Is there a way to turn it off?

Thanks and cheers!

axd, 2020-04-24:
What Linux distro, kernel version are you using for this?

foehnph, 2020-04-24:
I was using it on ubuntu 18.04, kernel version I don't remember.
I just updated to ubuntu 20.04 on 5.4.0-26-generic. also **working** without problems!

axd, 2020-05-12:
Thanks @foehnph, I'm also using Ubuntu 20.04, but for some reason I can't build the module and therefore load it to the kernel, could you please share your installation process and steps?

I got the follow errors:

```
a@ov:~/Gits/xmm7360-pci$ make && make load
make -C /lib/modules/5.4.0-29-generic/build M=/home/user/Gits/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-29-generic'
  CC [M]  /home/user/Gits/xmm7360-pci/xmm7360.o
/home/user/Gits/xmm7360-pci/xmm7360.c: In function ‘xmm7360_remove’:
/home/user/Gits/xmm7360-pci/xmm7360.c:1177:6: warning: unused variable ‘i’ [-Wunused-variable]
 1177 |  int i;
      |      ^
/home/user/Gits/xmm7360-pci/xmm7360.c: In function ‘xmm7360_probe’:
/home/user/Gits/xmm7360-pci/xmm7360.c:1375:6: warning: unused variable ‘i’ [-Wunused-variable]
 1375 |  int i, ret;
      |      ^
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/user/Gits/xmm7360-pci/xmm7360.mod.o
  LD [M]  /home/user/Gits/xmm7360-pci/xmm7360.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-29-generic'
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Operation not permitted
make: *** [Makefile:19: load] Error 1

```

foehnph, 2020-05-12:
actually, form your output it looks like it built just fine, but it didn't load.

after building it with `make`, I typically load it using a separate command:
`sudo make load`    *(<-- tl:dr try that)*
compared to your `make && make load`
I think it should ask you for sudo authorization anyway, but probably make sure your shell is authenticated.
btw.: The first `rmmod: ERROR` is expected if the module is not yet loaded. it does no harm.

axd, 2020-05-12:
Wow, that really did the trick. I was unable to get this running before, now I'm connected. I hope to make this work with ModemManager. 

Thanks again for your help. :-)

closed 2021-03-10 by tgxn