xmm7360/xmm7360-pci #25: X1 Carbon 7

alexsosic, 2020-02-01:
Hi there,

first of all thanks for the work done, much appreciated. I'm trying to run it with the following command:
```sudo python3 rpc/open_xdatachannel.py --apn internet```
and in the output I find this line:
```INFO:root:Attach failed - waiting to see if we just weren't ready```

dmesg gives this output:
```[  294.088028] xmm7360: loading out-of-tree module taints kernel.
[  294.088147] xmm7360: module verification failed: signature and/or required key missing - tainting kernel
[  294.104187] xmm7360 0000:02:00.0: enabling device (0000 -> 0002)
[  294.105079] xmm7360 0000:02:00.0: modem is ready
```

`ip a` gives this output:
```4: wwan0: <POINTOPOINT,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/none ```

The link is down and if I try to add a New connection in the Advanced Network Configuration if does not recognize the wwan0 as a mobile broadband modem so I cannot associate it.
As I understand by your documentation the python command should already connect the modem and get an IP and configure the route.

Any other things I should try?

fredr, 2020-02-06:
Do you have a pin code on your sim? In that case, see https://github.com/xmm7360/xmm7360-pci/issues/5#issuecomment-573305213 and https://github.com/xmm7360/xmm7360-pci/pull/21/files#diff-04c6e90faac2675aa89e2176d2eec7d8R18

ryzokuken, 2020-02-20:
Hi @alexsosic! I'm curious: did you make any progress here?

woopstar, 2020-02-24:
I have it working on my X1 Carbon gen 7.

```
$ alias wan_online='cd /home/user/xmm7360-pci; make; make load; nmcli radio wifi off; sudo python3 rpc/open_xdatachannel.py -r --apn www.internet.mtelia.dk -m 1000'
```

ryzokuken, 2020-03-04:
Mine fails with `ERROR:root:Attach failed again, giving up`

samtech09, 2020-04-23:
It's working for me, i just followed as instructed in Readme.
Thanks for the work.

mattmon, 2020-11-04:
I'm also having the same problem described by @alexsosic.

X1 7th Gen running debian testing, kernel 5.9.1

Anyone have any tips on getting it working?

pdme, 2021-02-11:
same issue here on X1 8th Gen running Arch

tgxn, 2021-03-09:
bump any progress here @alexsosic ; @mattmon ?

tgxn, 2021-03-23:
Closing stale issue, there's been a lot of repo changes so it would be ideal to re-test.

closed 2021-03-23 by tgxn