xmm7360/xmm7360-pci #27: Lenovo T590 issues (ERROR:root:Attach failed again, giving up)

GregLXStang, 2020-02-09:
On my T590 with the L850-GL WWAN, I've tried this to get it to work.  Make && make load gives me this:

```
make && make load
make -C /lib/modules/5.4.17-200.fc31.x86_64/build M=/home/ghager/Downloads/xmm7360-pci modules
make[1]: Entering directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/5.4.17-200.fc31.x86_64'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
```

Seems like all is well, but when I run the next command, it just gives me the below error:

```
sudo python rpc/open_xdatachannel.py --apn VZWINTERNET
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x3, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x2
FCC lock: state 1 mode 2
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0x1, 0x1e0, 0x137, 0x0
response: 0x0, 0xffffffff, 0x0, 0x1, 0xe, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0xe, 0x0
ERROR:root:Attach failed again, giving up
```

Unfortunately, I don't understand what's going on here...so I'm coming to you with them to try to get some help! Thank you in advance!

nlincke, 2020-02-24:
I have a similar problem. I get this message when i do the python command twice... Have you found a way to resolve this? One option is to restart this server...

frizop, 2020-05-19:
Seeing this fail as well, although my carrier is responding slightly differently:

```
RPC executing UtaMsNetAttachReq
b'__removed__'
async_ack: 0x0
unsolicited: UtaMsNetIsAttachAllowedIndCb: 0x0, 0x0, 0x0, 0x0
response: 0x0, 0xffffffff, 0x3, 0x3, 0x4, 0x0, 0x0, 0x0, 0x0, 0x0, 0x3, 0x0, 0x3, 0x4, 0x0
```

samweisgamdschie, 2021-03-14:
Hi, I had the same issue a few weeks ago and still run into this message `ERROR:root:Attach failed again, giving up` with the latest commits. I am running opensuse tumbleweed with linux kernel 5.11.2-1-default on an hp elitebook 840 G7, just FYI.

I followed the install steps with `make && make load`, entered the pin with the given command (which is needed with my SIM) and executed the python script as described. BUT I still wonder how this script know the parameters for my APN (a1.net) just by entering it's domain. Usually (e.g. in NM) there are plenty other parameters to set, like a Number, Username, and some authentication settings. I guess they differ between APNs?!

tgxn, 2021-03-25:
I've only had to set the APN for my modem, though I don't have  SIMPin enabled.
I don't know enough about how it gets those additional details, if they are required at all.

Are you able to try to disable the SIM Pin and try again if that is an option?

ge0rg, 2021-03-25:
It looks like the modem doesn't actually implement the `AT+CPIN` command, or maybe only accepts it once after power-up. I've run an interactive session (`minicom -D /dev/ttyXMM1` and then `ATE1` to enable echo), and it won't accept the PIN command:

```
AT+CPIN=?
OK
AT+CPIN?
ERROR
AT+CPIN="0815"        (this is not my real PIN)
ERROR
```

However, even after disabling the PIN on the SIM (which isn't so much of a good idea anyway), the script ends up emitting `ERROR:root:Attach failed again, giving up` until you **shutdown and restart the machine**.

After the restart, it finally worked and is providing ~30Mbit/s of downstream and ~15Mbit/s of upstream bandwidth, which is not too much but probably a limitation of the LTE network.

samweisgamdschie, 2021-03-28:
Many thanks, @ge0rg! debugging the modem with the above commands shows a similar picture.

So I tried to "disable" my SIM pin. But I am not sure how to do that. I put it into my smartphone and tried to change it. At first I tried to set no pin number at all but my smartphone declined it claiming to set at least a number. After that I set it to 0000 which my phone accepted. Is this the way to "disable SIM PIN"?

I debugged it again and at least got following output:

```
ATZ
OK
AT+CPIN=?
OK
AT+CPIN="0000"
OK
```
With the last command trying to enter my original PIN it returned `ERROR`. So I am a step further.

And, same here after reboot my machine && executing 
```
make load
echo "AT+CPIN=\"0000\"" | sudo tee -a /dev/ttyXMM1
python3 rpc/open_xdatachannel.py --apn my.apn.net
```

I finally got an IP. I had a trouble with DNS and resolved it by setting all existing network connections to state DOWN (disabling ethernet, wlan, etc.) and reexecuting the last `pyhton3` command. After that WWAN with XMM7360 start working. I reached 134MBIt/s DOWN and 33MBit/s UP at the speedtest. I would say IT WORKS ;) . **Many thanks for your work so far**! This is how I get rid of this ERROR. Now I need to set APN specific settings, for some kind of my company's fallback connection to our intranet. I already mentioned it at https://github.com/xmm7360/xmm7360-pci/issues/62.