xmm7360/xmm7360-pci #30: "No such file or directory: /dev/xmm0/rpc" after setting the interface down.

sudoforge, 2020-02-12:
```
➜ sudo python3 rpc/open_xdatachannel.py --apn h2g2
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 30, in <module>
    r = rpc.XMMRPC()
  File "/home/sudoforge/code/src/github.com/xmm7360/xmm7360-pci/rpc/rpc.py", line 18, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
```

sudoforge, 2020-02-12:
Notes (it's late, I'll try to make time to clean this up after I rest):

- The interface is gone
- Re-initializing the kernel `.ko` module does not resolve this
- Rebooting seems to work :man_shrugging:

mikail0, 2020-03-30:
Hi,

could you find a solution?
I have the same issue.

eralgua, 2020-04-20:
Error is observed at the time of awaking the system from sleep. I've also observed it when system was idle for more than 10-12 hrs.

sudoforge, 2020-04-25:
@eralgua can you also recreate by creating the interface, setting the interface `down`, and then trying to re-create the interface using the script?

jhagberg, 2020-09-11:
Got the same issue here after waking system from sleep.

massimoruoroch, 2021-01-25:
On Lenovo T15, when I have this error, suspending and resuming again the laptop solve the problem

tgxn, 2021-03-10:
We are aware of suspend/resume issues #1 
Rebooting does seem to fix this... Otherwise booting to Windows also seems to fix.

I have some log files that I want to look into further to see if we can resolve this under Linux, but we'll track it in that issue.

closed 2021-03-10 by tgxn

dominikpegler, 2021-08-08:
I am experiencing the same issue now with X1 6G and Manjaro. Still no idea what has changed in my system, but a reboot didn't solve it. I am currently not able to use the modem.

tgxn, 2021-08-09:
@dominikpegler  what about [this](https://github.com/xmm7360/xmm7360-pci/issues/99#issuecomment-872781874)

    `make unload`
    `systemctl suspend`
    resume
    `make load`

dominikpegler, 2021-08-09:
this didnt help, after ```make && make load``` I got the following output:

```rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Invalid module format
make: *** [Makefile:19: load] Error 1```

But after I ran ```make clean``` the make and make load commands suceeded. I now added ```make clean``` at the beginning of my shell script. which looks now like this currently:

```
make clean &&
make && make load &&
make && make load &&
sudo python ./rpc/open_xdatachannel.py || sudo python ./rpc/open_xdatachannel.py && sudo ip link set wwan0 up
```