xmm7360/xmm7360-pci #31: Offical Linux Support

nlincke, 2020-02-24:
First: Nice Work!!!

I have not been successfull running this fix but anyway...

Has anyone news about the official driver support? Or the ModemManager Integration?

enzingerm, 2020-03-04:
What exactly did not work for you with this solution?

I can't say anything about official driver support, but I am currently working on the ModemManager integration for this driver. It is done by writing a custom plugin for the ModemManager which takes care of the RPC communication which I have ported from Python to C. I just now was able to connect to the internet with the L850-GL modem and ModemManager/NetworkManager successfully for the first time (however, disconnecting/signal reporting/reasonable error handling is not yet built in). But it's still very hacky in many regards, as I had no experience with ModemManager before this.
However, I am planning to publish it in a ModemManager fork in the next days/weeks and maybe ask for help of other people in extending the functionality.

hectorakelsoe, 2020-03-14:
@enzingerm Where can we follow your work?

nexus166, 2020-03-14:
@enzingerm please let us know! I'd also love to help out: can develop and test on a T495.

enzingerm, 2020-03-14:
I pushed my fork to https://github.com/enzingerm/ModemManager
It works for me with a T495 and the L850-GL modem.
Almost all code is in `src/plugins/xmm7360` (other than that, I only changed something in `mm-filter.c` to allow the exposed tty port of the modem and the Makefiles. As I said, even basic features as disconnecting a connected bearer is not yet implemented and there are for sure numerous bugs hidden. So try it on your own risk, just load the kernel module and start ModemManager. (`ModemManager --debug`) might be helpful.

sparks-st1x, 2020-03-28:
Hi, Dell published this "https://dl.dell.com/FOLDER05569884M/1/L850_vV6.1.0.7_Linux.deb" tool to manage firmware of xmm7360 aka Fibocom L850 aka DW5820e. BTW I have dell 5401 and it's possible via BIOS to select default mode for the WWAN PCI/USB

tazhate, 2020-04-06:
@enzingerm how can we say "thanks" in dollars for you work? donate url will be awesome.

enzingerm, 2020-04-07:
@tazhate Thank you very much for your offer. However, I don't feel like accepting donations for that project, as 1. it is still in a very pre-alpha state and 2. it is based heavily on the great work of @abrasive And as you might have seen, unfortunately there were no commits in the last few weeks. I still plan to further work on the ModemManager integration but there are other projects around which consume my time. A more important question for me is, did somebody actually get the ModemManager to work? I could provide further information on how to get the project running, if there is demand for that.

enzingerm, 2020-04-07:
> Hi, Dell published this "https://dl.dell.com/FOLDER05569884M/1/L850_vV6.1.0.7_Linux.deb" tool to manage firmware of xmm7360 aka Fibocom L850 aka DW5820e, I assume it will be useful
> for you. BTW I have dell 5401 and it's possible via BIOS to select default mode for the WWAN PCI/USB

Actually I'm not quite sure what this package does. At first glance, it seems to provide some kind of a firmware update utility for Linux, but according to the udev rules it assumes the modem to be connected via USB.

RealTehreal, 2020-04-21:
> First of all, I'm glad that there are actually people interested in getting this to work +1
> 
> I pushed my fork to https://github.com/enzingerm/ModemManager
> It works for me with a T495 and the L850-GL modem (please note that I use a SIM with disabled PIN, I don't know if it works if you have a PIN set).
> Almost all code is in `src/plugins/xmm7360` (other than that, I only changed something in `mm-filter.c` to allow the exposed tty port of the modem and the Makefiles. As I said, even basic features as disconnecting a connected bearer is not yet implemented and there are for sure numerous bugs hidden. So try it on your own risk, just load the kernel module and start ModemManager. (`ModemManager --debug`) might be helpful.

Thank you so much for your effords to make this work. I'm using a T495, too. I'm on Pop!OS 19.10 and now I wonder, what I actually have to do. I took a look at your ModemManager fork but as far as I can see it is described, what it is. But not what I have to do to make it work.
I would love to have just a little small bit of advice on how to actually use it. Thank you so much in advance.

enzingerm, 2020-04-22:
@RealTehreal I added a README with instructions with my latest commit. I am sure that it is not complete at the moment (for example I didn't remember all dependencies I installed to be able to build the ModemManager project). And just because it works on my system doesn't mean it will work flawlessly on yours. But I'd appreciate it if you share your experience when you tried it, maybe by creating an issue or so.

RealTehreal, 2020-04-25:
Hey, thanks for the response. Sorry for answering late, I'm quite busy right now.
I hopped over to your ModemManager fork and will continue there. Already got you an issue ;-)

tdemeter, 2020-07-14:
I can confirm, this driver works on Lenovo X1 Yoga Gen3 (latest bios) with Fedora 32 latest kernel.

Please do not give up! It's a very nice work and very useful for a lot of people.

wicadmin, 2020-07-18:
Yeah, please don't give up....hoping to see the xmm7560 supported too.

tgxn, 2021-03-24:
Related project/work: https://github.com/enzingerm/ModemManager/tree/xmm7360/plugins/xmm7360

lu0, 2021-06-17:
> I can confirm, this driver works on Lenovo X1 Yoga Gen3 (latest bios) with Fedora 32 latest kernel.
> 
> Please do not give up! It's a very nice work and very useful for a lot of people.

Hi! Were you able to send/receive SMS? Were you talking about the ModemManager plugin?

tgxn, 2021-06-21:
> > I can confirm, this driver works on Lenovo X1 Yoga Gen3 (latest bios) with Fedora 32 latest kernel.
> > Please do not give up! It's a very nice work and very useful for a lot of people.
> 
> Hi! Were you able to send/receive SMS? Were you talking about the ModemManager plugin?

Hey, I don't believe anyone has managed to get SMS working via this device/driver combo.

tgxn, 2021-08-01:
Uhhm, https://support.lenovo.com/au/en/downloads/ds550634-fibocom-l860-gl-wireless-wan-driver-for-linux-thinkpad-x1-yoga-gen-4-x1-carbon-gen-7-x1-yoga-gen-5-x1-carbon-gen-8

?
Maybe i'm just tired. test tomorrow.

tgxn, 2021-08-02:
Figured I'd link this thread here since it's related to official support: https://github.com/intel/linux-intel-lts/issues/7

enzingerm, 2021-08-05:
I just wanted to note that official Linux support for the 7560 modem [will be included in 5.14](https://www.phoronix.com/scan.php?page=news_item&px=Intel-M.2-Modem-Linux-5.14)
It seems that the general architecture is not that different but the 7560 exposes MBIM directly over one of its shared memory queue pairs, which the 7360 doesn't according to the author of the xmm7360-pci driver. I have fiddled around with it a bit (included the 7360 to the list of supported PCI device IDs) and at least it doesn't crash. However, I can't communicate with the provided AT ports. So I think there is still some modification to be made to support the 7360 properly.

jan-kiszka, 2021-08-05:
Thanks for the heads up. Already asked on LKML the Intel developer if 7360 could benefit from that driver as well? I would assume he can answer this.

jan-kiszka, 2021-08-05:
Anyway, just did that: https://lore.kernel.org/netdev/0545a78f-63f0-f8dd-abdb-1887c65e1c79@siemens.com/T/#u

enzingerm, 2021-08-05:
I just asked by sending a direct mail to him, but obviously using the LKML is the better and more visible approach.

jan-kiszka, 2021-08-05:
Yeah, sometimes this works as well, but sometimes you are just being ignored. I usually deal with direct emails by asking for the reason for not asking (again) publicly. Best case, you get a searchable answer for the archives.

tgxn, 2021-08-09:
FWIW, the 7560 modem [seems so close the a simple s/7360/7560/ makes this driver compatible.](https://github.com/xmm7360/xmm7360-pci/pull/50)

nlincke, 2021-08-09:
Hey guys, its incredible that are still updates to this thread! And its even more promising that we might get a out of the box solution!
I am still running the USB Switch script.... But i would love to get rid of this.... 

I will cross my fingers! ;)

enzingerm, 2021-08-12:
I just (again) stumbled upon a thread in the ModemManager mailing list, where in 2018 (!) a Fibocom engineer wanted advice on how to integrate ports created over PCIe into ModemManager (targeting L860-GL/XMM7560). Considering [this](https://lists.freedesktop.org/archives/modemmanager-devel/2018-December/006954.html) message, as I understand it the fundamental difference between L850-GL and L860-GL is that the former doesn't expose MBIM over PCIe (which is not new, but lacks definitive confirmation from e.g. Intel). 

So even if the new `iosm` driver could be adapted to work for the L850-GL, we would still be left with using the non standard RPC language so the only benefit would be to have better power management handling (support suspend/resume?) as well as hopefully a more stable PCIe driver in general.

I'm quite disappointed by the lack of support by Intel/Fibocom. At the moment I'm trying to hopefully get my modem exchanged with a L860-GL. This is because I can't use the USB mode switch on my AMD based T495 (USB device won't be recognized). But I don't even know if the L860-GL is whitelisted in the BIOS so that I will be able to use it at all.

blackandcold, 2021-08-18:
> But I don't even know if the L860-GL is whitelisted in the BIOS so that I will be able to use it at all.

I tried two different modems in the T14, neither was recognised. Hope is low but wish you the best.

A shame for such companies to not support Unix drivers. The change from USB to PCIe-only is the real problem with the 850 in my device.. Fibocom not releasing any information or driver is showing how little they care at all.
I'd love to use LTE with my notebook but it is unreliable so I have to use Windows on the go or waste my phone power...

golyalpha, 2022-07-11:
Apparently, support for the modem has been added in Linux 5.18 - it detects fine, and shows up in nmcli and `ip a`, but I still can't get it to work - reports sim-missing even though SIM is clearly inserted in the laptop's SIM slot, and works fine under Windows (I have already unlocked the modem in Windows)

jan-kiszka, 2022-07-11:
I vaguely recall it takes some Network Manager changes to make use of the upstream driver. Check the related kernel thread when it was proposed for merging. I didn't find the time to dig deeper into that myself, though it would be great to have it Just Working (TM).

golyalpha, 2022-07-11:
My problem with the linux kernel driver right now, is that the modem doesn't see the SIM for some reason, and fails to initialize as a result - a problem that is absent when I reboot into Windows, where that same modem works fine. I tried removing the PIN from the SIM card, but that didn't help. I do have a feeling that this is less of a kernel driver problem, and more of a ModemManager problem though.

wicadmin, 2022-07-11:
@golyalpha  - are you able to turn on debug in MM and paste the log or open a ticket on their gitlab?

golyalpha, 2022-07-12:
Installed the modem yesterday, and need to do other things first. But will do as soon as I get around to it.

jirutka, 2022-07-12:
> Apparently, support for the modem has been added in Linux 5.18

Can you please post a link to the source of this information? I found only mentions of 7560, but not 7360.

hinaloe, 2022-07-12:
it.
https://github.com/torvalds/linux/commit/1f52d7b622854b8bd7a1be3de095ca2e1f77098e

wicadmin, 2022-07-12:
> ~Installed the modem yesterday, and need to do other things first. But will do as soon as I get around to it.~
> 
> Just checked the ModemManager repo, seems like my issue has probably already been fixed in ModemManager, but Fedora is a little behind with their ModemManager package.

You should be able to install MM itself outside of the distrubution versions. I do not believe (but could be wrong) there is really any tight dependencies. modemmanger, libmbin, libqmi and a few others.

joostpjotr, 2022-07-12:
I've tried with Linux 5.18.5 and a ModemManager build from the `main` branch at `f2a3d836`, and also get the _SIM not inserted_ error. This is on a Lenovo ThinkPad X1 Nano Gen 1. Alas, I can't rule out some issue with the SIM card or tray, because I don't have Windows available and I didn't try of any of the other ways to get this device working.

Here is a log of the AT-commands issued by ModemManager: [wwan0.log](https://github.com/xmm7360/xmm7360-pci/files/9094453/wwan0.log)

abrasive, 2022-07-12:
This sounds like it could be the Fibocom "FCC lock" feature, which keeps 
the modem stuck in flight mode until their driver authenticates.
See the FCC lock section here for details: 
https://github.com/xmm7360/xmm7360-usb-modeswitch#fcc-lock

joostpjotr, 2022-07-12:
> This sounds like it could be the Fibocom "FCC lock" feature, which keeps the modem stuck in flight mode until their driver authenticates.

I've looked into this, and it seems an additional, but separate issue. I gather that you can query the _phone functionality level_ using the `AT+CFUN?` command with level 1 meaning _fully functional_ and level 4 meaning _flight mode_. Initially the modem is in flight mode, and trying `AT+CFUN=1` to bring it up does not change that.

However, using the instructions at https://github.com/xmm7360/xmm7360-usb-modeswitch#fcc-lock it seems possible to bring the radio up:

[image]

golyalpha, 2022-07-14:
@hinaloe thanks for that one, that is indeed the commit that alerted me to support being added

@wicadmin 

True, it's more that I just don't like installing things that are ordinarily managed by the system package manager manually.

@prometheanfire 

Interesting - what was the output of the fibocom/intel/xmm plugins when dealing with that device? Maybe support in MM is still not properly implemented?

@joostpjotr 

Which plugin picked up the wwan0at1 device? intel or generic? If generic then you're probably having the same issue as I do.

And yes, if you don't have Windows on your computer, you are likely to have FCC Lock still on. (Well, not anymore after unlocking - seems you're lucky to have old enough firmware to do it this way)

golyalpha, 2022-07-17:
I guess we're still not quite there yet with ModemManager support... 

As for the eSIM - yes, it has to be obtained from the network operator, afaik they should issue you some kind of a "token", that you somehow load into the eSIM that makes it work as that operator's SIM. 

Unfortunately, I don't have access to the customer center with the SIM I currently have in my laptop and would want to use with the modem, and don't feel like using my other SIM in my phone to test this, only to go through the hassle of cancelling the eSIM and getting a new physical SIM for the phone... Especially if it's not exactly certain that all this will work.

davidovitch, 2022-07-21:
> This sounds like it could be the Fibocom "FCC lock" feature, which keeps the modem stuck in flight mode until their driver authenticates. See the FCC lock section here for details: https://github.com/xmm7360/xmm7360-usb-modeswitch#fcc-lock

I am no expert here, but it seems things go wrong before that. Purely based on the log output it seems that it first complains the sim is not inserted:

```
ModemManager[60264]: <debug> [71.604431] [modem0/wwan0at1/at] --> 'AT+CGDCONT=?<CR><LF>'
ModemManager[60264]: <debug> [71.606896] [modem0/wwan0at1/at] <-- '<CR><LF>+CME ERROR: 10<CR><LF>'
ModemManager[60264]: <debug> [71.606951] [modem0/wwan0at1/at] operation failure: 10 (SIM not inserted)
ModemManager[60264]: <warn>  [71.606978] [modem0] couldn't load supported IP families: SIM not inserted
```
and only then it tries to check if it is locked, but it fails since it couldn't detect a sim:

```
ModemManager[60264]: <debug> [71.609919] [modem0/wwan0at1/at] --> 'AT+CPIN?<CR><LF>'
ModemManager[60264]: <debug> [71.612827] [modem0/wwan0at1/at] <-- '<CR><LF>+CME ERROR: 10<CR><LF>'
ModemManager[60264]: <debug> [71.612922] [modem0/wwan0at1/at] operation failure: 10 (SIM not inserted)
ModemManager[60264]: <debug> [71.612959] [modem0] couldn't check if unlock required: SIM not inserted
ModemManager[60264]: <info>  [71.613004] [modem0] state changed (unknown -> locked)
ModemManager[60264]: <warn>  [71.613110] [modem0] modem couldn't be initialized: Couldn't check unlock status: SIM not inserted
```
This is on a HP Elitebook 845 G7 with
* Linux 5.18.11-arch1-1 # 1 SMP PREEMPT_DYNAMIC, (Arch Linux)
* ModemManager 1.18.10
* L850 LTE Module 18500.5001.00.03.25.23.

brenns10, 2022-07-30:
I'm not really familiar with the inner workings of SIM cards, but I was playing around trying to get the `iosm` module (plus ModemManager) to work on my 5.19 kernel and noticed something odd. On a fresh boot, if I run ModemManager with a valid (and tested on another device) SIM card inserted, I get the same result everyone is getting here (SIM not inserted). However, if I then do the following steps, I get a different error message:

1. `rmmod iosm`
2. Build and load the xmm7360 module
3. Use the `rpc/open_xdatachannel.py` script to connect (and this works great!)
4. `rmmod xmm7360`
5. `modprobe iosm`
6. Now, run `ModemManager` again!

Here's the error message from the fresh boot:

```
ModemManager[6002]: <info>  ModemManager (version 1.18.10-2) starting in system bus...
ModemManager[6002]: [qrtr] socket lookup from 1:0
ModemManager[6002]: [qrtr] initial lookup finished
ModemManager[6002]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:02.2/0000:02:00.0': not supported by any plugin
ModemManager[6002]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:02.3/0000:03:00.0': not supported by any plugin
ModemManager[6002]: <info>  [device /sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0] creating modem with plugin 'generic' and '3' ports
ModemManager[6002]: <warn>  [plugin/generic] could not grab port wwan0at0: Cannot add port 'wwan/wwan0at0', unhandled port type
ModemManager[6002]: <info>  [base-manager] modem for device '/sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0' successfully created
ModemManager[6002]: <warn>  [modem0] couldn't load supported IP families: SIM not inserted
ModemManager[6002]: <info>  [modem0] state changed (unknown -> locked)
ModemManager[6002]: <warn>  [modem0] modem couldn't be initialized: Couldn't check unlock status: SIM not inserted
ModemManager[6002]: <info>  [modem0] state changed (locked -> failed)
```

And here's the error message I get after loading xmm7360, unloading it, and switching back to iosm:

```
ModemManager[7933]: <info>  ModemManager (version 1.18.10-2) starting in system bus...
ModemManager[7933]: [qrtr] socket lookup from 1:0
ModemManager[7933]: [qrtr] initial lookup finished
ModemManager[7933]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:02.2/0000:02:00.0': not supported by any plugin
ModemManager[7933]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:02.3/0000:03:00.0': not supported by any plugin
ModemManager[7933]: <info>  [device /sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0] creating modem with plugin 'generic' and '3' ports
ModemManager[7933]: <warn>  [plugin/generic] could not grab port wwan0at0: Cannot add port 'wwan/wwan0at0', unhandled port type
ModemManager[7933]: <info>  [base-manager] modem for device '/sys/devices/pci0000:00/0000:00:02.5/0000:05:00.0' successfully created
ModemManager[7933]: <warn>  [modem0/sim0] couldn't load operator name: SIM failed to handle CRSM request (sw1 106 sw2 130)
ModemManager[7933]: <warn>  [modem0/sim0] couldn't load list of emergency numbers: Failed to parse CRSM query result '+CRSM: 105,129'
ModemManager[7933]: <warn>  [modem0/sim0] couldn't load list of preferred networks: Not found
ModemManager[7933]: <warn>  [modem0] couldn't load list of own numbers: Operation not allowed
ModemManager[7933]: <info>  [modem0] state changed (unknown -> disabled)
```

I haven't seen this error message in this bug before, so I wonder if the `xmm7360` module does some initialization on the SIM card that ModemManager doesn't do with `iosm`, and after using `xmm7360`, ModemManager can load further? Honestly just hoping somebody who knows more might find this useful.

SebSK3, 2022-08-12:
Confirming driver iosm still not working on 18500.5001 firmware. "sim-missing" error. Thinkpad T495s. 

Good thing we have this project. For anyone having problems i was following this https://github.com/xmm7360/xmm7360-pci/pull/181 and managed to make it work

flokli, 2022-08-15:
This is probably the more appropriate place to discuss this: https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/612

nursoda, 2023-01-08:
I documented the current state as good as I could in https://wiki.archlinux.org/title/Xmm7360-pci#Linux_Kernel_support_and_Modem_Manager_integration