xmm7360/xmm7360-pci #35: Can't connect to wwan0: Device class NMDeviceGeneric had no complete_connection method

jeryini, 2020-04-17:
Hello!

Network interface `wwan0` is successfully created with IP assigned:
```wwan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 100.64.21.213  netmask 0.0.0.0  destination 100.64.21.213
        inet6 fe80::3c9d:27bb:f56d:6301  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
I then followed [this](https://unix.stackexchange.com/questions/113975/configure-gsm-connection-using-nmcli) instruction on how to connect to `wwan0` interface. But unfortunately when I execute command to connect `sudo nmcli device connect wwan0` I get the following error:
```
Error: Failed to add/activate new connection: Device class NMDeviceGeneric had no complete_connection method
```
I'm not sure what I'm doing wrong. Thank you for your help.

VogtME, 2020-05-05:
I have this same issue! When I start this tool, I seem to get a working network interface. It even grabs the IP address it is supposed to according to the DNS scheme; however I cannot connect to it! I see that the connection is described here as up, but I cannot send data through it. 

`wwan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 35.24.195.115  netmask 0.0.0.0  destination 35.24.195.115
        inet6 fe80::f275:ee3b:c85f:11e1  prefixlen 64  scopeid 0x20<link>
        inet6 ::fcdb:20ee:7c52:dd7  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`
NMCLI Shows: 
`wwan0: unmanaged
        "Intel XMM7360 Advanced"
        unknown (xmm7360), hw, mtu 1500
`
I also have an X1 Carbon 6th Gen.dns

tgxn, 2021-03-23:
I have just merged NetworkManager support #66 so this may resolve your issues accessing it.
Otherwise, there was previously no support for nmcli, so this would have only just started working.

Given you are both getting IP Addresses from the modem connection, this would indicate you might have a related problems such as:
- WiFI Connection may need to be disabled
- You may need to remove/use the telco-provided DNS servers.
- You may need to check that no conflicting routes are present.

tgxn, 2021-03-25:
I'm going to close this issue, if you do continue to have issues, could I ask that you try the new `dbus` functionality, and create a new issues if this is still causing issues.

closed 2021-03-25 by tgxn