xmm7360/xmm7360-pci #36: Can not get it to work on T480 with arch

udithvb, 2020-04-19:


udithvb, 2020-04-19:
I had to install pyroute2 and linux headers(linux-headers 5.6.4.arch1-1) to build and execute , I get a wwan0 interface but no ip address.
I disabled sim card lock to avoid complications(reset it to 1234)
After setting wwan0 by using following command:
`sudo ip link set wwan0 up`
I get a inet6 address but I still cannot connect to the internet. 
Please help!

frizop, 2020-05-19:
That IPV6 address is a link local address (ie: it uses the mac and builds you a unique address based on local info)

If you set a pin, can you try the following: https://github.com/xmm7360/xmm7360-pci/pull/21/files#diff-04c6e90faac2675aa89e2176d2eec7d8R18

`echo "AT+CPIN=\"0000\"" | sudo tee -a /dev/ttyXMM1`. Replace `0000` with your pin code.

udithvb, 2020-05-23:
Thanks @frizop,  I did not miss this command, I even confirmed the pin on my phone. 
I only get an IPV6 address, but I can't browse the internet.

mikail0, 2020-07-26:
Hey,
I have a T480, too and it works for me.

I wrote a small guide:
    sudo pacman -S python-pip
    Ensure Python package ConfigArgParse is installed for root eg. sudo pip install --user ConfigArgParse
    sudo pip install --user pyroute2
    git clone https://github.com/xmm7360/xmm7360-pci.git
    cd in the directory to which you downloaded the repo
    Install kernel headers: sudo pacman -S linux-headers
    sudo make
    sudo make load (ignore error that module xmm7360 is not loaded)
    sudo python3 rpc/open_xdatachannel.py --apn your.apn.here
You should receive a wwan0 interface, with an IP, and a default route.
If the last command doesnt finish: Boot to windows, activate wwan and reboot to linux and run it all again.
WLAN must be turned off while running the last command.

udithvb, 2020-07-27:
Hey, thanks for the comment, however it did not work.

I am dumping stdout incase it is helpful, I noticed that it printed out the carrier, so I guess something is happening:
```
sudo python3 rpc/open_xdatachannel.py --apn jio.net 
```

```
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x20742b97
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c02040000019211000101020411000101020477e7014c'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaMsCpsSetModeIndCb: 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa0204000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x5, 0xf, 0x91, 0x2, 0x0, 0x2, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x3, 0x0, 0x90, 0x3, 0x1, 0x3, 0x1, 0x0, 0x0

```

```
4: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none
    **inet6 fe80::3665:6a2c:4ed9:6f4c/64** scope link stable-privacy
       valid_lft forever preferred_lft forever
```

I did disable wlan and enable only wwan using rfkill.

mikail0, 2020-07-27:
I dont understand why it is not working for you, sorry.
Maybe it is because of your network carrier. As you can see in other threads here, some carriers are not working.

frizop, 2020-09-03:
> Same here, unfortunately.
> 	
> device 	Lenovo T490
> kernel 	`5.8.1-arch1-1`
> os 	`arch`
> carrier 	Google Fi
> APN 	`h2g2`
> card 	`2c:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)`

@boz can you confirm that the SIM card you're using in your T490 works in a phone as a data only sim? I ask because I'm using google Fi, on a T480, or a T14 and have not been able to get this to work. I did test two different sims in a phone, and google Fi was able to pick them up no problem and use data.

boz, 2020-09-03:
@frizop I haven't used it on a phone, but I'll give it a try ASAP and let you know.

I'd like to tinker with the protocol handling here but I don't really know where it's documented at all.   If anyone has any pointers I'd appreciate it.

blackandcold, 2020-09-21:
> T14 and have not been able to get this to work. 

Hi there, T14 user here. Got it finally to work. I updated all drivers and FW in Windows and Linux, removed the pin in Windows and then did the chore like https://github.com/xmm7360/xmm7360-pci/issues/36#issuecomment-664031236 described:

`❯ git clone https://github.com/xmm7360/xmm7360-pci.git
Klone nach 'xmm7360-pci' ...
remote: Enumerating objects: 323, done.
remote: Counting objects: 100% (323/323), done.
remote: Compressing objects: 100% (158/158), done.
remote: Total 323 (delta 203), reused 281 (delta 161), pack-reused 0
Empfange Objekte: 100% (323/323), 74.65 KiB | 967.00 KiB/s, Fertig.
Löse Unterschiede auf: 100% (203/203), Fertig.
❯ cd xmm7360-pci
❯ sudo make
make -C /lib/modules/5.8.6-1-MANJARO/build M=/home/bac/GitRepositories/Privat/xmm7360-pci modules
make[1]: Verzeichnis „/usr/lib/modules/5.8.6-1-MANJARO/build“ wird betreten
  CC [M]  /home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.o
/home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.c: In Funktion »xmm7360_remove«:
/home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.c:1177:6: Warnung: Variable »i« wird nicht verwendet [-Wunused-variable]
 1177 |  int i;
      |      ^
/home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.c: In Funktion »xmm7360_probe«:
/home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.c:1375:6: Warnung: Variable »i« wird nicht verwendet [-Wunused-variable]
 1375 |  int i, ret;
      |      ^
  MODPOST /home/bac/GitRepositories/Privat/xmm7360-pci/Module.symvers
  CC [M]  /home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.mod.o
  LD [M]  /home/bac/GitRepositories/Privat/xmm7360-pci/xmm7360.ko
make[1]: Verzeichnis „/usr/lib/modules/5.8.6-1-MANJARO/build“ wird verlassen
❯ sudo make load
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Fehler 1 (ignoriert)
sudo /sbin/insmod xmm7360.ko
`

❯ ip link show wwan0
5: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 1000
    link/none 

status unknown but internet works. Prepaid Austrian SIM here.

What I have to do is disconnect WIFI and then run the connection script.

Need to reproduce this again.

XenGi, 2020-10-16:
Same problem here. I could get it to load the driver and the wwan0 interface would show up but not more then that.

But the situation got way worse for me. I booted into windows and installed all the updates that it wanted. Now back under Arch Linux, when I do a `make load` this happens:

```
$ make load
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
make: *** [Makefile:19: load] Segmentation fault (core dumped)
```

The kernel log shows this and after that the kernel crashes completely.

```
[   66.370171] xmm7360 0000:02:00.0: modem is ready
```

I'm not sure whats wrong. Did I update to an unsupported version of the firmware?

frizop, 2020-10-16:
Sounds like an exception is not caught? Curious

XenGi, 2020-10-17:
I switched to a Fibocom L830-EB and it is recognized as a USB device instantly. Tools like modemmanager and mbimctl seem to work with it too. I have to wait for my antenna cables to arrive to check if I can get any signal but it looks promising.
So for everyone in europe/africa/asia the L830-EB is definitly the better option. AFAIK only the L850 does this PCI fuckery.

Peter-Ries, 2020-12-30:
I'd change my modem immediately if only there wasn't this Whitelist in T14 BIOS ... :/

At least I got my modem to work with this driver. :+1:

tgxn, 2021-03-23:
Hey,

There's been a few repo changes that could resolve these problems, if you're still having problems or made any progress on this please let me know; otherwise I'll close this issue.

zhuyifei1999, 2021-03-23:
Works for me. Gentoo Linux + NetworkManager + Google Fi.

closed 2021-03-23 by tgxn