xmm7360/xmm7360-pci #37: Broken with kernel 5.6.7-arch1-1

samtech09, 2020-04-28:
Initially it was working well with `kernel 5.6.4` on my *Lenovo X1 Carbon 7th gen*, but after upgrading to latest `kernel 5.6.7`, it is now broken.
`make` is successful, but `make load` is giving following error

```
$ make load
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not insert module xmm7360.ko: Invalid module format
make: *** [Makefile:19: load] Error 1
```

samtech09, 2020-04-28:
Hi, ignore this issue, it is actually working. What i have to do is to clean previous build files. Following is working

```
make clean
make && make load
```

closed 2020-04-28 by samtech09

mbernasocchi, 2020-09-01:
I had the same and had to disable secure boot in the bios