xmm7360/xmm7360-pci #40: HP EliteBook 830 G5

SergeyDjam, 2020-06-18:
Hi. Would i like to know what status of EliteBook support? Can I check my laptop for modem health and how do I do it?

xentity, 2020-07-22:
hey @SergeyDjam,
did you make any progress? I have an HP EliteBook 840 G5 and had no success so far. The hack of @abrasive seems not to work for HP EliteBooks (for any kind of reason), Lenovo and Dell are more promising. What are your experiences so far?

RyanShaw-3083, 2020-08-04:
Hi @SergeyDjam , I have already compiled and load module on Elitebook 735G6. It actually works! Amazoning! Thanks a lot to @abrasive focus on this field!

bear80, 2020-11-02:
Hi @SergeyDjam , just loaded the module on Elitebook 850 G6. Works fine! Thanks a lot to @abrasive !

tgxn, 2021-03-09:
I'll be looking to merge MM support soon! 🎉 https://github.com/xmm7360/xmm7360-pci/pull/66

closed 2021-03-09 by tgxn

kolAflash, 2021-04-07:
Works on an EliteBook 735 G6 (9VZ68EA#ABD / Ryzen 5 PRO 3500U) with a upgraded "Fibocom L850-GL 4G LTE" wwan card.  

The EliteBook 735 G6 doesn't come with a wwan card in every variant.  
For example my EliteBook 735 G6 9VZ68EA#ABD (Ryzen 5 PRO 3500U) doesn't have a wwan card.  
So I upgraded it with a wwan card bought here:  
https://www.ebay.de/itm/Fibocom-Anatel-L850-GL-4G-LTE-WWAN-Modul-M-2-HP-P-N-917822-002-/402030952348  
https://www.notebookgalerie.de/fibocom%20l850-gl%204g%20lte%20wwan-modul%20m.2.html  

---

Entering the sim pin seemed a little problematic.  
I think I maybe have to run `open_xdatachannel.py ...` once before entering the pin as described here: https://github.com/xmm7360/xmm7360-pci/issues/53#issuecomment-770162791  
And entering the pin via picocom seemed to work better.
```
$ picocom /dev/ttyXMM1
AT+CPIN="0000"
```
Use `AT+CPIN?` to check if a pin is needed.  
Answer `+CPIN: SIM PIN` indicates that a pin is still needed.  
Answer `CPIN+: READY` indicates no pin is not needed or the pin has already been entered.

yummy527, 2022-09-09:
> Works on an EliteBook 735 G6 (9VZ68EA#ABD / Ryzen 5 PRO 3500U) with a upgraded "Fibocom L850-GL 4G LTE" wwan card. OS: Debian-11, Linux-5.10
> 
> The EliteBook 735 G6 doesn't come with a wwan card in every variant. For example my EliteBook 735 G6 9VZ68EA#ABD (Ryzen 5 PRO 3500U) doesn't have a wwan card. So I upgraded it with a wwan card bought here: https://www.ebay.de/itm/Fibocom-Anatel-L850-GL-4G-LTE-WWAN-Modul-M-2-HP-P-N-917822-002-/402030952348 https://www.notebookgalerie.de/fibocom%20l850-gl%204g%20lte%20wwan-modul%20m.2.html
> 
>  
> 
>  
> 
> Entering the sim pin seemed a little problematic. I think I maybe have to run `open_xdatachannel.py ...` once before entering the pin as described here: [#53 (comment)](https://github.com/xmm7360/xmm7360-pci/issues/53#issuecomment-770162791) And entering the pin via picocom or chat (Debian package `ppp`) seemed to work better.
> 
>  
> 
> ```
> $ picocom /dev/ttyXMM1
> AT+CPIN="0000"
> ```
> 
> Use `AT+CPIN?` to check if a pin is needed. Answer `+CPIN: SIM PIN` indicates that a pin is still needed. Answer `CPIN+: READY` indicates no pin is not needed or the pin has already been entered.
> 
>  
> 
> Alternative: `chat -Ss -V -t 1 '' 'AT+CPIN?' '+CPIN: SIM PIN' 'AT+CPIN="0000"' 'OK' 'AT+CPIN?' '+CPIN: READY' > /dev/ttyXMM1 < /dev/ttyXMM1` (don't get confused, the expected answers are listed too in the chat command)

The command AT+CPIN? gives me ERROR. What it means?