xmm7360/xmm7360-pci #42: T14 AMD - got wwan0 and ip - how to proceed

Peter-Ries, 2020-07-28:
Thanks a lot for taking care of things lenovo does not!!!

I finally got the module built on my brand new T14 with Ryzen. (Kubuntu 20.04 with mainline kernel 5.8.0.rc7)

I got around the problem of no wwan0 after resume with a reboot. Bringing the interface wwan0 down with "sudo ip link wwan0 down" does not work - fails with interface not found. 

So I now got a wwan0 interface with an ip address, but how can I go on? Internet does not work yet. Modem Manager integration ist not there yet.

Thanks for any hint.

BR
Peter

mcbridedm, 2020-08-08:
Did you try manually setting DNS servers?

blackandcold, 2020-11-18:
> So I now got a wwan0 interface with an ip address, but how can I go on? Internet does not work yet. Modem Manager integration ist not there yet.
> 
Hi, how did you do that?
Got the T14 AMD 4750U too.

I'm stuck at getting the interface up.

Any hints?

Thanks guys!

Peter-Ries, 2020-12-29:
Just a sum up after I took some time to dive into this topic again.

With the help of this script (posted somewhere in issues of this repo (https://github.com/wafflepie/dotfiles/blob/backup/fedora/devices/waffle-x13y/lte.sh) and a modification to telekom Germny and my paths I can now say

IT'S WORKING ON T14 AMD!!!

Just open a terminal and call "lte setup" the modem comes up (lot of output in terminal, but I left it open) and I have a working internet connection. "lte down" stops the modem.

For comfort a modemmanager integration would be nice but as I only have to use WWAN seldomly it's suficcient for me.

THANKS!

closed 2020-12-29 by Peter-Ries