xmm7360/xmm7360-pci #43: NetworkManager integration or bash script

diplatis, 2020-08-03:
Are there any plans integrating with NetworkManager?
Or, until then, any bash script/alias for connecting and disconnecting?
Thanks!

wafflepie, 2020-08-11:
Hi, here is a script that I use for connecting: https://github.com/wafflepie/dotfiles/blob/fedora/devices/waffle-x13y/lte.sh

Just change the paths to where you've cloned this repository (and the APN, obviously). After changing them and assuming you already got everything working, you can just run `./lte.sh setup` in the directory of the shell script. Next time, to connect, just run `lte` (assuming you have `.local/bin` in your `$PATH`). Sometimes I need to run it twice to get the DNS working, but other than that, no issues on my side (ThinkPad X13 Yoga LTE on Fedora 32). I had to remove the PIN from my SIM card in Windows, though.

(Sorry, I don't have a script for disconnecting, Wi-Fi seems to have a higher priority on my side automatically and I never need not to be connected to the internet.)

jhagberg, 2020-09-09:
Looks like the link is broken is this the correct file ? https://github.com/wafflepie/dotfiles/blob/master/devices/waffle-x13y/lte.sh

closed 2021-03-23 by tgxn