xmm7360/xmm7360-pci #54: Works on Lenovo T14s AMD Manjaro 20.1.2

yoxcu, 2020-10-24:
Managed to compile it after installing linux-headers and pyroute2. Using a Telecom SIM and disabled WLAN interface.
First execution throws a exception:

I```
NFO:root:IP address: 0.0.0.0, 0.0.0.0, 0.0.0.0
INFO:root:DNS server(s): 
0.0.0.0
0.0.0.0
0.0.0.0
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 89, in <module>
    address=ip_addr)
NameError: name 'ip_addr' is not defined

```
but waiting a few seconds the script runs fine and i get a wwan0 interface with an ip-adress along with a working connection!

Thanks for the awesome work!

yoxcu, 2020-10-24:
#26 solved the issue of executing the script twice
worked on both 5.8 and 5.9 kernel

closed 2021-03-09 by tgxn

ItanVazlak, 2021-06-27:
good day.

I am a novice Linux user. I could not independently install this driver on Lenovo T480s and Manjaro.

Can you help me and write here a sequence of commands in the terminal for Manjaro?

I have installed all the packages I need, but when I run the "make && make load" command, I get an error:

"make [1]: Entering directory '/usr/lib/modules/5.10.42-1-MANJARO/build'
make [1]: *** No rule to make target 'modules'. Stop.
make [1]: Leaving directory '/usr/lib/modules/5.10.42-1-MANJARO/build'
make: *** [Makefile: 9: default] Error 2
"

tgxn, 2021-06-27:
@ItanVazlak G'day, It looks like you might not have kernel headers installed, 

You could try an article like [this](https://linuxconfig.org/manjaro-linux-kernel-headers-installation) one.

yoxcu, 2021-06-27:
Hi,

make sure you have the (right version) of kernel-headers installed!

`sudo pacman -S linux-headers`
should do the trick!

If installing through pamac (gui) search for the the right version. For you it should be
`linux510-headers`
according to your log.