xmm7360/xmm7360-pci #63: Signal strength issue.

zhdanow5a, 2021-01-21:
Xmm2usb works well for me( as service), but i dont want to wait 30-40 sec for start working.
So i install xmm-pci and want to build gnome applet with signal strengh/status network... 
Unfortunately  xxmpci is not shown signal strength via AT commands while xmm2usb does: 
at+csq
+CSQ: 10,4
OK
xmm-pci:
 at+csq
 +CSQ: 99,99  ( Unknown or not detectable)
 OK
What can i do? Is  it can be changed via other at commands, or it is fault of native driver and only hard work with it can solve it issue.

tgxn, 2021-02-24:
I would guess that this is functionality that is only provided in the official windows driver.

It'd be nice if we supported this, but due to the nature of the undocumented API's I'm not holding out hope.

Ecos-hj, 2021-03-02:
I've done some integration work, mostly for setting the SIM pin. 
So I've figured out at the moment you somehow send AT commands via the device node the whole things gets very unstable. 
I had to made some wrappers arround the python script, but I ended up with disabling the SIM pin, to get things work properly. 
Most of the stuff I've done in perl, that's why I haven't done the commits.

tgxn, 2021-03-13:
Hey @Ecos-hj I think there's some specific unsolicited messages that come in, and you can see in the `rpc.py` script, the only message that we pay attention to (other than solicited replies) is the [UtaMsNetIsAttachAllowedIndCb](https://github.com/xmm7360/xmm7360-pci/blob/master/rpc/rpc.py#L112) message,

I'm having some issues on my laptop with establishing a connection, and I've seen some various other messages coming in (related to sim/signal/connection/net/etc) which could be decoded to mean something, though I'm not sure on the data format.

Ecos-hj, 2021-03-13:
I've spent a lot of the to get the modem work with an activated SIM in a productive and stable way. 
And in the end we should avoid to use this serial tty interface with AT commands at all. 
Yeah the information should be extracted out of the byte packets we get or sent from the rpc interface. 
I have no clue if the basic work with this was done by reverse engineering or if there is any kind of documentation around.

tgxn, 2021-03-13:
I'm fairly sure it was mostly RE. [This revering repo](https://github.com/xmm7360/reversing) seems to have more information and some IDA files, though I'm getting stuck at opening them. :)

tgxn, 2021-03-24:
Related project/work: https://github.com/enzingerm/ModemManager/tree/xmm7360/plugins/xmm7360

zhuyifei1999, 2021-03-25:
> I'm getting stuck at opening them. :)

IDA free 7.0 says database version too old. 

```
$ python pyidbutil/idbtool.py -i win_driver/*

==> win_driver/ModemAuthenticator.i64 <==

loader: pe64.llx64 Portable executable for AMD64 (PE)
cpu: metapc, version=695, filetype=0b:Portable Executable (PE), ostype=win, apptype=exe,64bit, core:0, size:0
idaver=0695: 6.95
nopens=9, ctime=5da50e08, crc=f8303588, md5=4710f3c026f690f4ab911d422952539f
orig: v0695 2016-10-05 22:11:46 .. ....-..-.. ..:..:..  48-3257-7594-82  James Laird-Wah
user: v0695 2016-10-05 22:11:46 .. 2017-03-23 09:00:00  48-3257-7594-82  James Laird-Wah

==> win_driver/ModemControl.i64 <==

loader: pe64.llx64 Portable executable for AMD64 (PE)
cpu: metapc, version=695, filetype=0b:Portable Executable (PE), ostype=win, apptype=dll,64bit, core:0, size:0
idaver=0695: 6.95
nopens=2, ctime=5da50f4f, crc=071fd0e6, md5=84fd7aa6363590709e3e6c191b3d3042
orig: v0695 2016-10-05 22:11:46 .. ....-..-.. ..:..:..  48-3257-7594-82  James Laird-Wah
user: v0695 2016-10-05 22:11:46 .. 2017-03-23 09:00:00  48-3257-7594-82  James Laird-Wah

==> win_driver/Ude.i64 <==

loader: pe64.llx64 Portable executable for AMD64 (PE)
cpu: metapc, version=695, filetype=0b:Portable Executable (PE), ostype=win, apptype=64bit, core:0, size:0
idaver=0695: 6.95
nopens=15, ctime=5d79a5be, crc=ca3824cd, md5=546c7a7ae3e590f7801ddcec94d17268
orig: v0695 2016-10-05 22:11:46 .. ....-..-.. ..:..:..  48-3257-7594-82  James Laird-Wah
user: v0695 2016-10-05 22:11:46 .. 2017-03-23 09:00:00  48-3257-7594-82  James Laird-Wah
$ python pyidbutil/idbtool.py -i firmware/*

==> firmware/18500.5001.00.02.22.93.fls_ID2_CODE_LoadMap0.idb <==

loader: None None
cpu: ARM, version=695, filetype=02:Binary File, ostype=, apptype=, core:0, size:169af84
idaver=0695: 6.95
nopens=17, ctime=5da90c8d, crc=ff6f1f1a, md5=2713db25e2d517abe649cf43facbcd71
orig: v0695 2016-10-05 22:11:46 .. ....-..-.. ..:..:..  48-3257-7594-82  James Laird-Wah
user: v0695 2016-10-05 22:11:46 .. 2017-03-23 09:00:00  48-3257-7594-82  James Laird-Wah
```

Seems to be IDA version 6.95. I don't find any freeware of this version. I got a friend who has a IDAPro license, will ask if he has 6.95.

tgxn, 2021-03-25:
@zhuyifei1999 Yep! I got exactly the same thing! I had tried with IDA 5 and 7.

I'm not sure what version these were created against or how we can get them open :')

zhuyifei1999, 2021-03-25:
Yep, he has 6.95, but "Memory map is taking forever to XML export. Here are the idcs meanwhile"
[firmware_sdc.zip](https://github.com/xmm7360/xmm7360-pci/files/6208303/firmware_sdc.zip)

I'll send a PR I guess when he has all of them exported. 

However, IDA free does not support ARM, which the firmware is on. (And I have no IDA license.) What's the best way to do this? Port to Ghidra?

zhuyifei1999, 2021-03-26:
> I'll send a PR I guess when he has all of them exported.

PR: https://github.com/xmm7360/reversing/pull/3

> Port to Ghidra?

https://github.com/zhuyifei1999/xmm7360-reversing/tree/ghidra. Function argument information seems bad. Probably needs a lot of manual reversing.