xmm7360/xmm7360-pci #68: Network Manager doesn't recognize the connection, refuses to connect to VPN

alembiq, 2021-02-11:
As i can't see wwan connection in the networkmanager, i can't use network manager to connect to VPN, as it's convinced there is no connection :( 

Parrot OS (Debian based)
Lenovo X1 gen 7

tgxn, 2021-02-24:
It is known that this PCI driver does not expose the network connection to NetworkManager.

There is a current PR open, https://github.com/xmm7360/xmm7360-pci/pull/66 that is meant to add support for this, however I have not been able to test it.

Perhaps you could check that code out and see if that functionality is working for you?

alembiq, 2021-03-02:
tested and working :)

closed 2021-03-23 by tgxn

deelerke, 2021-09-14:
I'm having the same issue.. is there a workaround? 
thanks