xmm7360/xmm7360-pci #80: Tried on Thinkpad L15

nicholascarroll, 2021-03-20:
Hi, I tried this on Thinkpad L15, but I don't know whether it worked or not. Following is the install log.

to-s, 2021-03-20:
On L14 Gen 1 it's working.

nicholascarroll, 2021-03-20:
nick@L15:~/Downloads$ cd ~/xmm7360-pci-master/
nick@L15:~/xmm7360-pci-master$ ls
DEVICES.md  Makefile  README.md  rpc  trace  xmm7360.c
nick@L15:~/xmm7360-pci-master$ sudo apt install build-essential python3-pyroute2 python3-configargparse git
[sudo] password for nick:                                                                                            
Reading package lists... Done                                                                                        
Building dependency tree                                                                                             
Reading state information... Done                                                                                    
build-essential is already the newest version (12.8ubuntu3).                                                         
build-essential set to manually installed.                                                                           
git is already the newest version (1:2.27.0-1ubuntu1.1).                                                             
git set to manually installed.                                                                                       
The following packages were automatically installed and are no longer required:
  libdirectfb-1.7-7 liblz1 libmimetic0v5 libqgpgme7 libqt5keychain1 ofono vim-runtime
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  python-pyroute2-doc
The following NEW packages will be installed:
  python3-configargparse python3-pyroute2
0 to upgrade, 2 to newly install, 0 to remove and 0 not to upgrade.
Need to get 284 kB of archives.
After this operation, 1,681 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://au.archive.ubuntu.com/ubuntu groovy/universe amd64 python3-configargparse all 0.13.0-2 [22.6 kB]
Get:2 http://au.archive.ubuntu.com/ubuntu groovy/main amd64 python3-pyroute2 all 0.5.14-0ubuntu1 [261 kB]
Fetched 284 kB in 9s (32.3 kB/s)          
Selecting previously unselected package python3-configargparse.
(Reading database ... 298605 files and directories currently installed.)
Preparing to unpack .../python3-configargparse_0.13.0-2_all.deb ...
Unpacking python3-configargparse (0.13.0-2) ...
Selecting previously unselected package python3-pyroute2.
Preparing to unpack .../python3-pyroute2_0.5.14-0ubuntu1_all.deb ...
Unpacking python3-pyroute2 (0.5.14-0ubuntu1) ...
Setting up python3-pyroute2 (0.5.14-0ubuntu1) ...
Setting up python3-configargparse (0.13.0-2) ...
nick@L15:~/xmm7360-pci-master$ ^C
nick@L15:~/xmm7360-pci-master$  make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
  CC [M]  /home/nick/xmm7360-pci-master/xmm7360.o
  MODPOST /home/nick/xmm7360-pci-master/Module.symvers
  CC [M]  /home/nick/xmm7360-pci-master/xmm7360.mod.o
  LD [M]  /home/nick/xmm7360-pci-master/xmm7360.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ sudo python3 rpc/open_xdatachannel.py --apn telstra.wap
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
[... 104 lines cut ...]
nameserver 8.8.8.8
nick@L15:~/xmm7360-pci-master$ echo "nameserver 9.9.9.9" | sudo tee -a /etc/resolv.conf
nameserver 9.9.9.9
nick@L15:~/xmm7360-pci-master$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 8c:8c:aa:4f:0d:5a brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether b0:7d:64:1e:77:65 brd ff:ff:ff:ff:ff:ff
5: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none 
    inet 244.31.1.74/0 scope global wwan0
       valid_lft forever preferred_lft forever
    inet6 2001:8004:1201:a375:eb59:e490:4aba:1ec7/64 scope global dynamic mngtmpaddr stable-privacy 
       valid_lft forever preferred_lft forever
    inet6 fe80::92a9:aa49:ba08:c2f7/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
nick@L15:~/xmm7360-pci-master$ ping google.com
^C
nick@L15:~/xmm7360-pci-master$

nicholascarroll, 2021-03-20:
<pre>
nick@L15:~/Downloads$ cd ~/xmm7360-pci-master/
nick@L15:~/xmm7360-pci-master$ ls
DEVICES.md  Makefile  README.md  rpc  trace  xmm7360.c
nick@L15:~/xmm7360-pci-master$ sudo apt install build-essential python3-pyroute2 python3-configargparse git
[sudo] password for nick:                                                                                            
Reading package lists... Done                                                                                        
Building dependency tree                                                                                             
Reading state information... Done                                                                                    
build-essential is already the newest version (12.8ubuntu3).                                                         
build-essential set to manually installed.                                                                           
git is already the newest version (1:2.27.0-1ubuntu1.1).                                                             
git set to manually installed.                                                                                       
The following packages were automatically installed and are no longer required:
  libdirectfb-1.7-7 liblz1 libmimetic0v5 libqgpgme7 libqt5keychain1 ofono vim-runtime
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  python-pyroute2-doc
The following NEW packages will be installed:
  python3-configargparse python3-pyroute2
0 to upgrade, 2 to newly install, 0 to remove and 0 not to upgrade.
Need to get 284 kB of archives.
After this operation, 1,681 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://au.archive.ubuntu.com/ubuntu groovy/universe amd64 python3-configargparse all 0.13.0-2 [22.6 kB]
Get:2 http://au.archive.ubuntu.com/ubuntu groovy/main amd64 python3-pyroute2 all 0.5.14-0ubuntu1 [261 kB]
Fetched 284 kB in 9s (32.3 kB/s)          
Selecting previously unselected package python3-configargparse.
(Reading database ... 298605 files and directories currently installed.)
Preparing to unpack .../python3-configargparse_0.13.0-2_all.deb ...
Unpacking python3-configargparse (0.13.0-2) ...
Selecting previously unselected package python3-pyroute2.
Preparing to unpack .../python3-pyroute2_0.5.14-0ubuntu1_all.deb ...
Unpacking python3-pyroute2 (0.5.14-0ubuntu1) ...
Setting up python3-pyroute2 (0.5.14-0ubuntu1) ...
Setting up python3-configargparse (0.13.0-2) ...
nick@L15:~/xmm7360-pci-master$ ^C
nick@L15:~/xmm7360-pci-master$  make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
  CC [M]  /home/nick/xmm7360-pci-master/xmm7360.o
  MODPOST /home/nick/xmm7360-pci-master/Module.symvers
  CC [M]  /home/nick/xmm7360-pci-master/xmm7360.mod.o
  LD [M]  /home/nick/xmm7360-pci-master/xmm7360.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ sudo python3 rpc/open_xdatachannel.py --apn telstra.wap
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
[... 106 lines cut ...]
nick@L15:~/xmm7360-pci-master$ echo "nameserver 9.9.9.9" | sudo tee -a /etc/resolv.conf
nameserver 9.9.9.9
nick@L15:~/xmm7360-pci-master$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 8c:8c:aa:4f:0d:5a brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether b0:7d:64:1e:77:65 brd ff:ff:ff:ff:ff:ff
5: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none 
    inet 244.31.1.74/0 scope global wwan0
       valid_lft forever preferred_lft forever
    inet6 2001:8004:1201:a375:eb59:e490:4aba:1ec7/64 scope global dynamic mngtmpaddr stable-privacy 
       valid_lft forever preferred_lft forever
    inet6 fe80::92a9:aa49:ba08:c2f7/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
nick@L15:~/xmm7360-pci-master$ ping google.com
^C
nick@L15:~/xmm7360-pci-master$ 
</pre>

tgxn, 2021-03-22:
So the current build doesn't have NetworkManager support yet, that's in PR #66.

```
INFO:root:IP address: 244.31.1.74
INFO:root:DNS server(s): 10.4.58.204, 10.4.130.164, 10.4.58.204, 2001:8004:0:a1a0:7:e1c0:0:10, 2001:8004:0:100:7:e1c0:0:14
RPC executing UtaMsCallPsConnectReq
```
This would indicate you have an active connection on `wwan0`.

Which you can see on `ip addr`:
```
5: wwan0:  mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none 
    inet 244.31.1.74/0 scope global wwan0
```

I'm not sure why you're not able to ping, are you sure:
- Your service on the sim is active?
- You might need to disable WiFi?
Otherwise, I'm connected and using my Telstra SIM right now.

tgxn, 2021-03-22:
It could also have something to do with your nameservers, what happens if you just try to use the Telstra-provided ones?

nicholascarroll, 2021-03-22:
nick@L15:~$ cd xmm7360-pci-master/
nick@L15:~/xmm7360-pci-master$ make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
[sudo] password for nick: 
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ make && make load
make -C /lib/modules/5.8.0-44-generic/build M=/home/nick/xmm7360-pci-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-44-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-44-generic'
sudo /sbin/rmmod xmm7360
sudo /sbin/insmod xmm7360.ko
nick@L15:~/xmm7360-pci-master$ cat /etc/resolv.conf
# Generated by NetworkManager
search iiNet
nameserver 127.0.0.53
nick@L15:~/xmm7360-pci-master$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 8c:8c:aa:4f:0d:5a brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000                                                                                        
    link/ether b0:7d:64:1e:77:65 brd ff:ff:ff:ff:ff:ff                                        
    inet 10.1.1.8/24 brd 10.1.1.255 scope global dynamic noprefixroute wlp3s0                 
       valid_lft 44138sec preferred_lft 44138sec                                              
    inet6 fe80::48ed:36e1:8575:80f4/64 scope link noprefixroute                               
       valid_lft forever preferred_lft forever                                                
5: wwan0: <POINTOPOINT,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/none                                                                                 
nick@L15:~/xmm7360-pci-master$ echo i have disabled wifi
i have disabled wifi
nick@L15:~/xmm7360-pci-master$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 8c:8c:aa:4f:0d:5a brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether b0:7d:64:1e:77:65 brd ff:ff:ff:ff:ff:ff
5: wwan0: <POINTOPOINT,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/none 
nick@L15:~/xmm7360-pci-master$ sudo python3 rpc/open_xdatachannel.py --apn telstra.wap
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
[... 83 lines cut ...]
2: enp2s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 8c:8c:aa:4f:0d:5a brd ff:ff:ff:ff:ff:ff
3: wlp3s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether b0:7d:64:1e:77:65 brd ff:ff:ff:ff:ff:ff
5: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none 
    inet 10.142.64.154/0 scope global wwan0
       valid_lft forever preferred_lft forever
    inet6 2001:8004:cc1:9b3f:9ea3:87dc:78fa:5fd2/64 scope global dynamic mngtmpaddr stable-privacy 
       valid_lft forever preferred_lft forever
    inet6 fe80::49cc:4b7e:119a:a9a1/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
nick@L15:~/xmm7360-pci-master$ ping google.com
PING google.com(syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e)) 56 data bytes
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=1 ttl=116 time=35.3 ms
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=2 ttl=116 time=43.7 ms
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=3 ttl=116 time=51.5 ms
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=4 ttl=116 time=49.4 ms
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=5 ttl=116 time=52.4 ms
64 bytes from syd15s03-in-x0e.1e100.net (2404:6800:4006:804::200e): icmp_seq=6 ttl=116 time=50.6 ms
^C
--- google.com ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5010ms
rtt min/avg/max/mdev = 35.295/47.128/52.353/5.985 ms
nick@L15:~/xmm7360-pci-master$

nicholascarroll, 2021-03-22:
GREAT IT DOES WORK!!!
Indeed @tgxn it was just a matter of leaving the name server as is.

so I am doing this:
`cd xmm7360-pci
make && make load
make && make load # doing it twice because the first time it has an error
sudo python3 rpc/open_xdatachannel.py --apn your.apn.wap
sudo ip link set wwan0 up
`
If there is not other bug to investigate here please close this.

THANKS SO MUCH FOR THIS DRIVER!

tgxn, 2021-03-22:
Hey @nicholascarroll No problems, happy to help you get it working!

I see what you mean that you had to run it twice, but the first time it is not an issue
`rmmod: ERROR: Module xmm7360 is not currently loaded` means the module wasn't loaded yet, which is fine since we're loading it for the first time.

tgxn, 2021-03-22:
I made a couple changes to the documentation to clarify installation and DNS, it should help clarify this as it has come up before. :)

closed 2021-03-22 by tgxn