xmm7360/xmm7360-pci #87: New device; Thinkpad X13 (Ryzen)

MerlijnWajer, 2021-03-30:
Working for me, stable connection for a few hours at least.

Thanks for the work.

MerlijnWajer, 2021-03-30:
OS is Gentoo, Kernel is 5.11.7.

tgxn, 2021-05-03:
Thanks for the feedback! I've added information for the X13.

closed 2021-05-03 by tgxn

embedded-solutions-art, 2021-10-12:
Hello folks :)
My device X13 AMD Ryzen7 with Ubuntu 21.04(5.11.0-37-generic Linux).
Unfortunately this driver not working on my PC. 

spietrza@think:~/lte/xmm7360-pci$ make
make -C /lib/modules/5.11.0-37-generic/build M=/home/spietrza/lte/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-37-generic'
  CC [M]  /home/spietrza/lte/xmm7360-pci/xmm7360.o
  MODPOST /home/spietrza/lte/xmm7360-pci/Module.symvers
  CC [M]  /home/spietrza/lte/xmm7360-pci/xmm7360.mod.o
  LD [M]  /home/spietrza/lte/xmm7360-pci/xmm7360.ko
  BTF [M] /home/spietrza/lte/xmm7360-pci/xmm7360.ko
**Skipping BTF generation for /home/spietrza/lte/xmm7360-pci/xmm7360.ko due to unavailability of vmlinux**
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-37-generic'
spietrza@think:~/lte/xmm7360-pci$ make load
sudo /sbin/rmmod xmm7360
**rmmod: ERROR: Module xmm7360 is not currently loaded**
**make: [Makefile:18: load] Error 1 (ignored)**
sudo /sbin/insmod xmm7360.ko
**insmod: ERROR: could not insert module xmm7360.ko: Operation not permitted**
**make: *** [Makefile:19: load] Error 1**
 
 Any advice?