xmm7360/xmm7360-pci #89: Works on ThinkPad T14

itkfm, 2021-04-09:
Hi,
thank you for your recent work on this. Especially the new helper script is very helpful.

I'm using Ubuntu 20.04.2 with the OEM-Kernel Linux <pcname> 5.6.0-1052-oem #56-Ubuntu SMP Fri Mar 19 15:56:22 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

I've tried kernel version 5.8 too and it works too.

merlinnot, 2021-04-19:
Works for me too on T14, Ubuntu 21.04, kernel 5.11.0-13-generic.

k1ngf15h3r, 2021-04-21:
I can not bring it up on Ubuntu LTS 20.04.2.0 with kernel 5.6.0.1055-oem. Getting ERROR:root:Attach failed again, giving up when trying "lte up" ... any hints or ideas?

merlinnot, 2021-04-21:
I'm no specialist here, but I had a similar error in the past. Do you have secure boot enabled? I do, in order to be able to install this module, you have to sign it.

I have a small script I run whenever I update my kernel, after recompiling the module:

```sh
sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der xmm7360.ko
hexdump -Cv xmm7360.ko | tail -n 5
```

To sign it, you need to have your MOKs registered. You should be able to find some guides online on how to do that.

Again, I'm not 100% sure that's the issue you are having. It just seems similar at a first glance.

k1ngf15h3r, 2021-04-21:
@merlinnot  thank you for the quick hint, but secure boot is disabled in my system. Did not have the MOK.priv and MOK.der files

zhuyifei1999, 2021-04-21:
FWIW,
>  [ 72.320276] xmm7360: module verification failed: signature and/or required key missing - tainting kernel

This actually means the module was loaded successfully. "tainting kernel" means the kernel records an 'untrusted module' has been loaded. This is more of a warning message than an error message.

k1ngf15h3r, 2021-04-21:
anyone with Telekom Deutschland SIM has success with this module? I tried APN internet.telekom and had set my PIN on each try

k1ngf15h3r, 2021-04-22:
I tried Ubuntu 20.04 LTS, Manjaro 20.1 and Fedora 33. Non of this systems works with this lte module

k1ngf15h3r, 2021-04-23:
Oh man ... after installing Ubuntu 21.04 I tried it with PIN disabled and now IT WORKS ;) thank you all

dadevel, 2021-04-24:
Hey @k1ngf15h3r I have a strange issue with my Deutsche Telekom SIM (PIN disabled). I can ping Googles DNS servers over IPv6 but not over IPv4. Furthermore the IPv4 address assigned to `wwan0` has a `/0` subnet which looks odd to me. I tried internet.telekom and internet.v6.telekom as APN. Both with the same result.

k1ngf15h3r, 2021-04-24:
> Hey @k1ngf15h3r I have a strange issue with my Deutsche Telekom SIM (PIN disabled). IPv6 works but IPv4 doesn't. For example I can ping Googles DNS servers over IPv6 but not over IPv4. Furthermore the IPv4 address assigned to `wwan0` has a `/0` subnet which looks odd to me. Also I noticed that `open_xdatachannel.py` adds only a IPv6 address to `/etc/resolv.conf`, no IPv4. I tried internet.telekom and internet.v6.telekom as APN. Both with the same result. 

Did you use the lte.sh script? I first used the rpc python script which connects but does not add the nameserver to /etc/resolv.conf. 

With "lte up" everything works fine (with Ubuntu 21.04)

I set "internet.telekom" as apn in the config INI file

tgxn, 2021-05-03:
Hey all! Looks like there's been lots of success in this thread. I've added the Lenovo T14 tot he devices list, with a link back to this thread.

closed 2021-05-03 by tgxn

Oualid33, 2021-05-25:
> Works for me too on T14, Ubuntu 21.04, kernel 5.11.0-13-generic.
Hi. which WWAN card is installed on your T14?

merlinnot, 2021-05-25:
Fibocom L850-GL 4G LTE CAT9.

Oualid33, 2021-05-29:
> Fibocom L850-GL 4G LTE CAT9.
Thank you for the reply.
My T14 came without WWAN module, so I've ordered a **Fibocom L-850 LG FRU/01AX792** as recommended by Lenovo on their forums. I wonder if this driver supports such a hardware?
Thank you very much.
NB: I am not specialist but I like using Linux.

alexbenisch, 2022-11-24:
ThinkPad T14 Gen 3 (14" Intel) 
Archlinux - Experience?
Can anyone using the fibocom l850-gl on this Hardware with Archlinux give a short description how to install and configure the 4g modem. 
Are there any problems with black listing, had that problem in the past? 
Maybe this will help others in the same situation.
Thank you all.