xmm7360/xmm7360-pci #90: Not working on ThinkPad P14s AMD

anu-devco, 2021-04-13:
Thanks for all your hard work, and I see a lot of people have gotten it to work but unfortunately I was unable to get it to work on Ubuntu 20.04

I have used your template scripts as well as executing commands manually, the script gets stuck on 

`sudo python3 rpc/open_xdatachannel.py`

there is no error and logs are similar to #27 but I don't get to the point where it says `ERROR:root:Attach failed again, giving up`

So I press ctrl + c to exit, I tried multiple times repeating the installation setups, and also restarted my system few times for making sure I get a fresh system.

anu-devco, 2021-04-13:
```
anu@thinkpad:~/tmp/xmm7360-pci$ sudo python3 rpc/open_xdatachannel.py --apn airtelgprs.com
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0xe385edbd
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c0204000001921100010102041100010102042b6fd477'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaMsCpsSetModeIndCb: 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 0xe, 0x2, 0x2, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x5, 0xf, 0x91, 0x2, 0x0, 0x2, 0x2, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x3, 0xb, 0x91, 0x3, 0x2, 0x3, 0x2, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x1, 0xff, 0x26, 0x0, 0x90, 0x4, 0xb, 0x4, 0xb, 0x0, 0x0

```

anu-devco, 2021-04-13:
On second run, FCC lock: state 1 mode 2

Not sure which is the right state

datagrams, 2021-04-19:
Hi @anu-devco,
I got my P14s AMD today and managed to get it to work without any issues, just went by the INSTALLING.md.

But I'm running Ubuntu 21.04 dev branch on 5.11.0-14-generic

anu-devco, 2021-04-19:
Oh see, thanks for telling me that. 

May I know which SIM you are using, I don't have much idea about logs generated if its related to my SIM, I have dual boot, and its works fine on windows but not on Ubuntu 20.04 with 5.6 linux kernal.  

Will it be possible if you can write the exact steps here, and also paste the logs generated for you, it could help me figure out the issue.

anu-devco, 2021-04-19:
Nevermind, I figured the out my issue. 
I am not sure 5.6 kernel as issue or not, as currently I am using 5.10.0-1022-oem kernel, so it works on this for sure.

**ACTUAL ISSUE : You should not be connected to any WIFI network, it can be ON but there shouldn't be any active connection. 
( Didn't tested with LAN if this behaviour is same on LAN )**

I used INSTALLING.md for first time setup. 

To connect to WAN, 
I am running following commands, and it works normally.
```
cd ~/tmp/xmm7360-pci/
sudo ./scripts/lte.sh setup
lte up
```

I would want to skip settting up WAN everytime I restart, just simply running **lte up** command would be great, and even better if someone can add can check for network connection to the **lte up** so it exits safely if an active connection is already there.

PS : If your **lte up**  commands get stuck even once, system restart it required, it happens if you accidently run it with an active WIFI connection

closed 2021-04-20 by anu-devco