xmm7360/xmm7360-pci #91: Works on T480s

mircsicz, 2021-04-20:
In my case on Fedora 33 after adding the following packages:

sudo dnf install python3-pyroute2 python3-configargparse
sudo dnf groupinstall "Development Tools" "Development Libraries"

Thank you for the effort :-*

tgxn, 2021-05-03:
Thanks for your feedback @mircsicz ! :)

mircsicz, 2021-05-14:
Keeps working after upgrading to FC34

But as with 33 I had DNS Problem's, what fixed it for me was adding:
`
resolvectl dns wwan0 8.8.8.8 8.8.4.4
`
to lte.sh like this:

`
if [[ "$1" = "up" ]]; then
  echo "bringing wwan0 up!" 

  python3 $SCRIPT_DIR/../rpc/open_xdatachannel.py
  ip link set wwan0 up
  resolvectl dns wwan0 8.8.8.8 8.8.4.4
fi
`

tgxn, 2021-05-14:
Funnily enough, I started experiencing DNS-related issues since Monday. It was failing to add any nameservers from the sims DHCP result not `/etc/resolv.conf` despite the configuration having it enabled.

I would rather build in support to use the provided nameservers (from sim) rather than forcing google's DNS on people. (Though I think this approach could work if we ran the `resolvectl dns wwan0 ns1 ns2` within the python script.