xmm7360/xmm7360-pci #96: issue with make on arch

brumle80, 2021-05-03:
❯ sudo make
make -C /lib/modules/5.12.1-arch1-1/build M=/home/dag/src/xmm7360-pci modules
make[1]: Entering directory '/usr/lib/modules/5.12.1-arch1-1/build'
  CC [M]  /home/***/src/xmm7360-pci/xmm7360.o
/home/dag/src/xmm7360-pci/xmm7360.c: In function ‘xmm7360_create_tty’:
/home/dag/src/xmm7360-pci/xmm7360.c:1166:13: error: ‘struct tty_port’ has no member named ‘low_latency’
 1166 |     qp->port.low_latency = 1;
      |             ^
make[2]: *** [scripts/Makefile.build:271: /home/***/src/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1851: /home/***/src/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.12.1-arch1-1/build'
make: *** [Makefile:9: default] Error 2
❯ uname -a
Linux 5.12.1-arch1-1 #1 SMP PREEMPT Sun, 02 May 2021 12:43:58 +0000 x86_64 GNU/Linux

zhuyifei1999, 2021-05-03:
Caused by https://github.com/torvalds/linux/commit/0bc1bd092af3c7c0b025ece93c3a86916f89f3ca. Drop it?

firun, 2021-05-09:
Has the exact same issue building on 5.12.2 on Ubuntu 21.04.

brumle80, 2021-05-18:
I guess this will be an issue for more people as the kernel gets updated to a certain version for more distros

mine is as of right now:

linux 5.12.4.arch1-2

brumle80, 2021-05-18:
and if I remove that referenced line it compiles and works correctly :)

tgxn, 2021-05-24:
> and if I remove that referenced line it compiles and works correctly :)

Do you mean [this line](https://github.com/xmm7360/xmm7360-pci/blob/master/xmm7360.c#L1332)?

tgxn, 2021-05-24:
I've commented the `low_latency` line in https://github.com/xmm7360/xmm7360-pci/blob/master/xmm7360.c#L1332 Maybe this could be tested on the new Kernel version?

DerChaot, 2021-05-31:
I can confirm, that this branch fixes the build issue on 5.12 using Manjaro and I can still connect using the WWAN.

LEXUGE, 2021-06-07:
Somehow on 5.12 and 5.10 with that PR, I am experiencing kernel panics.

Ecos-hj, 2021-06-22:
maybe this would be safer:

diff --git a/xmm7360.c b/xmm7360.c
index c5cbfa6..c92fde8 100644
--- a/xmm7360.c
+++ b/xmm7360.c
@@ -41,6 +41,7 @@
 #include <linux/wait.h>
 #include <linux/workqueue.h>
 #include <net/rtnetlink.h>
+#include <linux/version.h>
 
 MODULE_LICENSE("Dual BSD/GPL");
 
@@ -1329,7 +1330,9 @@ static int xmm7360_create_tty(struct xmm_dev *xmm, int num)
        struct queue_pair *qp = xmm7360_init_qp(xmm, num, 8, 4096);
        int ret;
        tty_port_init(&qp->port);
+#if (LINUX_VERSION_CODE<KERNEL_VERSION(5,12,0))
        qp->port.low_latency = 1;
+#endif
        qp->port.ops = &xmm7360_tty_port_ops;
        qp->tty_index = xmm->num_ttys++;
        tty_dev = tty_port_register_device(&qp->port, xmm7360_tty_driver,

flokli, 2021-07-04:
@Ecos-hj thanks for the patch, I opened https://github.com/xmm7360/xmm7360-pci/pull/117 to include this to the master branch.

tgxn, 2021-07-05:
Thanks @Ecos-hj - I much prefer that approach of checking for the user's kernel.

@LEXUGE I know you tested #101 and found an issue where you saw Kernel panics? If you could validate this branch and if you are still seeing errors let us know with a screenshot?

@m-kuhn and @DerChaot I know #101 worked for you so I assume there will be no issues with this new PR.

closed 2021-07-19 by tgxn