xmm7360/xmm7360-pci #100: Script fails with new Kernel Release

roald-fs, 2021-05-22:
Hello everybody,

I've been using the Lenovo T14 with Ryzen processor and OpenSuse Tumbleweed for 6 months. Most recently with Kernel 5.11. The Fibocomm Modem 850 gl ran perfectly with the kernel and the script! Thanks for that. Since changing the kernel to 5.12, I get the following error message when calling the script "./scripts/lte.sh setup".

Script: /lib/modules/5.12.4-1-default/xmm7360-pci-master/scripts/lte.sh; Link: / usr / local / bin
Requirement already satisfied: pyroute2 in /usr/local/lib/python3.8/site-packages (0.5.14)
Requirement already satisfied: configargparse in /usr/lib/python3.8/site-packages (1.3)
make -C /lib/modules/5.12.4-1-default/build M = / lib / modules / 5.12.4-1-default / xmm7360-pci-master modules
make [1]: Directory "/usr/src/linux-5.12.4-1-obj/x86_64/default" is entered
  CC [M] /lib/modules/5.12.4-1-default/xmm7360-pci-master/xmm7360.o
/lib/modules/5.12.4-1-default/xmm7360-pci-master/xmm7360.c: In function »xmm7360_create_tty«:
/lib/modules/5.12.4-1-default/xmm7360-pci-master/xmm7360.c:1332:10: Error: "struct tty_port" has no element named "low_latency"
 1332 | qp-> port.low_latency = 1;
      | ^
make [3]: *** [/usr/src/linux-5.12.4-1/scripts/Makefile.build:272: /lib/modules/5.12.4-1-default/xmm7360-pci-master/xmm7360 .o] Error 1
make [2]: *** [/usr/src/linux-5.12.4-1/Makefile:1865: /lib/modules/5.12.4-1-default/xmm7360-pci-master] Error 2
make [1]: *** [../../../linux-5.12.4-1/Makefile:215: __sub-make] Error 2
make [1]: The "/usr/src/linux-5.12.4-1-obj/x86_64/default" directory is exited
make: *** [Makefile: 9: default] Error 2
sudo / sbin / rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile: 18: load] Error 1 (ignored)
sudo / sbin / insmod xmm7360.ko
insmod: ERROR: could not load module xmm7360.ko: No such file or directory
make: *** [Makefile: 19: load] Error 1

Is there any experience or solutions?

It continues to run fine with the 5.11 kernel
Many Thanks!

brumle80, 2021-05-22:
Workaround:

delete line 1332 in the source code containing "qp->port.low_latency = 1;"
in xmm7360.c

Had to do that on my arch install, but it worked fine afterwards.

roald-fs, 2021-05-22:
tx a lot! your suggestion works fine.
T14s works with 
Operating System: openSUSE Tumbleweed 20210520
KDE Plasma Version: 5.21.5
KDE Frameworks Version: 5.82.0
Qt Version: 5.15.2
Kernel Version: 5.12.4-1-default
OS Type: 64-bit
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 PRO 4750U with Radeon Graphics
Memory: 30.6 GiB of RAM
Graphics Processor: AMD RENOIRwith opensuse tumbleweed and

roald-fs, 2021-05-22:
issue is solved

closed 2021-05-22 by roald-fs