xmm7360/xmm7360-pci #102: Trying to build kernel module on debian: "No Rule to make target modules"

solonlost, 2021-05-24:
I'm not terribly experienced with compiling things myself. I've tried to find out what's wrong with google, but the hits I'm getting don't appear to be related. I simply don't know what I'm doing wrong here:
```
user@system:/tmp/xmm7360-pci$ make && make load
make -C /lib/modules/5.10.0-0.bpo.5-amd64/build M=/tmp/xmm7360-pci modules
make[1]: Entering directory '/usr/lib/modules/5.10.0-0.bpo.5-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.10.0-0.bpo.5-amd64/build'
make: *** [Makefile:9: default] Error 2
```
I'm aware that it's probably something basic and stupid, but I've been stuck for a few days now and have no idea how to proceed, so any help would be appreciated.

I'm attempting to install on a T490s, running debian stable with 5.10 kernel from backports.

solonlost, 2021-05-31:
With debian again, I have the identical problem with an seeed odyssey J4125 sbc.

nivhty, 2021-06-02:
I'm having the same problem on Arch Linux 5.12.4-arch1-2 on Thinkpad X1 Gen 6th.

`make -C /lib/modules/5.12.4-arch1-2/build M=/tmp/Git/xmm7360-pci modules
make[1]: Entering directory '/usr/lib/modules/5.12.4-arch1-2/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.12.4-arch1-2/build'
make: *** [Makefile:9: default] Error 2
`

solonlost, 2021-06-07:
This was apparently a matter of not having the headers package installed. Installing:

`linux-headers-5.10.0-0.bpo.5-amd64`

package made it possible to run the build.

closed 2021-06-07 by solonlost

nivhty, 2021-06-09:
I have installed linux headers but still running into this issue.
```warning: linux-headers-5.12.9.arch1-1 is up to date -- **reinstalling**
resolving dependencies...
looking for conflicting packages...

Packages (1) linux-headers-5.12.9.arch1-1

Total Installed Size:  129.25 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n]```