xmm7360/xmm7360-pci #104: What's missing for mainline integration

jan-kiszka, 2021-05-31:
Just gave this a try on my Lenovo P52, and it work - great!

What all is missing to propose the driver for mainline linux? The low-hanging fruits are the 41 checkpatch warnings, but there might be more fundamental todos. Is the driver interface already considered stable enough?

The benefit of proposing it upstream would be attracting further reviews and improvement suggestions. And it could also raise more attention in the ModemManager, ofono etc. communities.

Oualid33, 2021-07-26:
I think suspend/resume is not currently supported.

jan-kiszka, 2021-07-27:
While that would be nice, I don't think that this alone would be a blocker. If the hardware cannot support saving and restoring the internal state across suspend/resume, the driver has no chance to provide that feature. But it should probably signal userspace that the device state changed.

My concerns would rather target the ABI to userland, if that is already in a state that could be frozen.

I've just started some trivial cleanups. But that revealed that the license situation is fuzzy - needs to be clarified first (#124).

tgxn, 2021-08-01:
Mainline proposes several issues, mostly that this driver is not stable, and has been reverse-engineered from the Windows driver.

This means that to support things like suspend/resume (or adding/fixing most driver issues), we would need someone to continue [the work by the original author](https://github.com/xmm7360/reversing), and build on the knowledge we have of this device. _(or hope official drivers get released 😫)_

Personally, I don't see the point of mainlining the driver since you'd still need to clone this repo and run the setup/config scripts anyway, but that's just my opinion. There's already several distros that link to this repo:
- https://wiki.archlinux.org/title/Xmm7360-pci

jan-kiszka, 2021-08-01:
Mainline is the easiest way intro distros. While some carry out-of-tree drivers but resistance is usually higher against such a model. Furthermore, mainline is the best way to involve other communities (such as modem-manager) as it signals stability, of the code and specifically of its interfaces. You also get quite a good code review (might be helpful, given that this driver can still freeze the machine).

I was in contact with experts at Lenovo / IBM (yeah, they still do support services) 2 years ago about getting an official driver for the PCI modem of my P52. From that I would say chances are minimal that this will ever happen.

tgxn, 2021-08-02:
Yeah, I agree it's highly unlikely that there will be any official driver released for this device from any manufacturer.

Happy for someone to run with this if they think it's worthwhile.

jan-kiszka, 2021-08-02:
I'll be happy to help, but someone else than me will have to explain the details of this design.

enzingerm, 2021-08-05:
Mainline support for the 7560 (not 7360) is to be added by Intel (see https://github.com/xmm7360/xmm7360-pci/issues/31#issuecomment-893756780)