xmm7360/xmm7360-pci #106: opensuse 15.2 on a Thinkpad t495

gariac, 2021-06-01:
I verified this works under win10. 

The wwan0 is never "up".  The state is unknown. I rolled everything back so that you can see a fresh installation.
---------------------------------
linux-yxjo:/usr/local/src/xmm7360-pci # make
make -C /lib/modules/5.3.18-lp152.75-default/build M=/usr/local/src/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-5.3.18-lp152.75-obj/x86_64/default'
  CC [M]  /usr/local/src/xmm7360-pci/xmm7360.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/local/src/xmm7360-pci/xmm7360.mod.o
  LD [M]  /usr/local/src/xmm7360-pci/xmm7360.ko
make[1]: Leaving directory '/usr/src/linux-5.3.18-lp152.75-obj/x86_64/default'
linux-yxjo:/usr/local/src/xmm7360-pci # make load
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
linux-yxjo:/usr/local/src/xmm7360-pci # python3 rpc/open_xdatachannel.py 
RPC executing UtaMsSmsInit
b'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
----------------------------------------------------------------
I have no idea if the number between the quotes is personally identifiable so I X'd it out.
Note at this point the shell is busy. Command prompt does not return. 

In another terminal:
sudo ip link set wwan0 up   
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 98:fa:9b:6e:3c:73 brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 98:fa:9b:6e:3c:72 brd ff:ff:ff:ff:ff:ff
4: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 5c:87:9c:70:90:4f brd ff:ff:ff:ff:ff:ff
    inet 192.168.43.206/24 brd 192.168.43.255 scope global dynamic noprefixroute wlan1
       valid_lft 3197sec preferred_lft 3197sec
    inet6 2607:fb90:27da:56c4:24ed:4ed:3331:b4f8/64 scope global temporary dynamic 
       valid_lft 3441sec preferred_lft 3441sec
    inet6 2607:fb90:27da:56c4:895a:d9d1:b834:55/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 3441sec preferred_lft 3441sec
    inet6 fe80::d94b:12e1:c527:ebf0/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
6: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/none 

This is the only device created.
/dev/xmm0> ls -l
total 0
crw------- 1 root root 240, 1 Jun  1 00:48 rpc
crw------- 1 root root 240, 3 Jun  1 00:48 trace
-------------------------------------------------------------------------------------------------------

Suggestions?

gariac, 2021-06-02:
Additional diagnostics added.

lspci:

`
 cat lspci_out 
04:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
        Subsystem: Intel Corporation Device 0020
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 77
        Region 0: Memory at d0800000 (64-bit, non-prefetchable) [size=4K]
        Region 2: Memory at d0801000 (64-bit, non-prefetchable) [size=1K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/4 Maskable- 64bit+
                Address: 00000000fee00000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L1, Exit Latency L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Not Supported
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [148 v1] Latency Tolerance Reporting
                Max snoop latency: 1048576ns
                Max no snoop latency: 1048576ns
        Capabilities: [150 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: xmm7360
        Kernel modules: xmm7360
`
dmesg:

`
dmesg |grep xmm7360
[    6.195173] xmm7360: loading out-of-tree module taints kernel.
[    6.251438] xmm7360: module verification failed: signature and/or required key missing - tainting kernel
[    6.251982] xmm7360 0000:04:00.0: enabling device (0000 -> 0002)
[    6.266350] xmm7360 0000:04:00.0: modem is ready
`

rfkill:
`
sudo rfkill
[sudo] password for root: 
ID TYPE      DEVICE                   SOFT      HARD
 0 bluetooth tpacpi_bluetooth_sw unblocked unblocked
 1 wwan      tpacpi_wwan_sw      unblocked unblocked
 2 wlan      phy0                unblocked unblocked
`

gariac, 2021-06-02:
Ack I accidentally closed it. I'm going to repost the diagnostics. The "code" feature of github doesn't work like other forums I have used and renders the text as a blob. 
========
lspci:
04:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
        Subsystem: Intel Corporation Device 0020
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 77
        Region 0: Memory at d0800000 (64-bit, non-prefetchable) [size=4K]
        Region 2: Memory at d0801000 (64-bit, non-prefetchable) [size=1K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/4 Maskable- 64bit+
                Address: 00000000fee00000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L1, Exit Latency L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Not Supported
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [148 v1] Latency Tolerance Reporting
                Max snoop latency: 1048576ns
                Max no snoop latency: 1048576ns
        Capabilities: [150 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: xmm7360
        Kernel modules: xmm7360
========

dmesg:

dmesg |grep xmm7360
[    6.195173] xmm7360: loading out-of-tree module taints kernel.
[    6.251438] xmm7360: module verification failed: signature and/or required key missing - tainting kernel
[    6.251982] xmm7360 0000:04:00.0: enabling device (0000 -> 0002)
[    6.266350] xmm7360 0000:04:00.0: modem is ready

===============

rfkill:
sudo rfkill
[sudo] password for root: 
ID TYPE      DEVICE                   SOFT      HARD
 0 bluetooth tpacpi_bluetooth_sw unblocked unblocked
 1 wwan      tpacpi_wwan_sw      unblocked unblocked
 2 wlan      phy0                unblocked unblocked
mrg@linux-yxjo:~/modemmanager>

tgxn, 2021-08-02:
@gariac 
Can you include the full output of `open_xdatachannel`?  The end result of this script will indicate if the connection failed or succeeded. The only PII that it'll contain are: Your IP Address and DNS Servers, and your chosen APN.

Your other debug information all looks normal, so I'm not sure what's going wrong.

tgxn, 2021-08-02:
Typical other things to check:
- Disable WiFi
- Disable SIM PIN
- Test device in windows/sim in a mobile phone

gariac, 2021-08-02:
On Sun, 01 Aug 2021 18:02:30 -0700
Domenic Horner ***@***.***> wrote:

> Typical other things to check:
> - Disable WiFi
> - Disable SIM PIN
> - Test device in windows/sim in a mobile phone
> 

The simcard is fine. I have it in a USB modem at the moment. It works
find in win10 on the T495.

gariac, 2021-08-02:
On Sun, 01 Aug 2021 18:00:54 -0700
Domenic Horner ***@***.***> wrote:

> @gariac
> Can you include the full output of `open_xdatachannel`?  The end
> result of this script will indicate if the connection failed or
> succeeded. The only PII that it'll contain are: Your IP [...]

At this point I will update the T495 to 15.3 before I waste any of your
time.

tgxn, 2021-08-02:
No problems at all, the full log can help, especially knowing the last messages that were sent/received from the modem before the startup script hangs for you.

It's not exactly the most stable driver, and does have some common problems around suspend/resume (#1) and the ability for the  included scripts (`open_xdatachannel.py`) from finishing. Updating sounds like a good plan 👍