xmm7360/xmm7360-pci #107: Works on a Lenovo X395

bencord0, 2021-06-01:
Device: Thinkpad X395
OS: Gentoo Linux (kernel version 5.10.27-gentoo)
Works: yes - Following the instructions in the README

I needed to `emerge dbus-python` on my system in addition to `pip install --user configargparse pyroute2` too, other distros may already have this installed.

This system is similar to the X390, except for the AMD processor and Radeon instead of Intel graphics.
`x86_64 AMD Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx AuthenticAMD GNU/Linux`

---
```
bencord0@localhost ~/src/xmm7360-pci $ lspci -k -d 8086:7360
03:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)
        Subsystem: Intel Corporation XMM7360 LTE Advanced Modem
        Kernel driver in use: xmm7360
bencord0@localhost ~/src/xmm7360-pci $ dmesg |grep xmm7360
[   68.796886] xmm7360: loading out-of-tree module taints kernel.
[   68.797527] xmm7360 0000:03:00.0: enabling device (0000 -> 0002)
[   68.797765] xmm7360 0000:03:00.0: modem is ready
bencord0@localhost ~/src/xmm7360-pci $ rfkill 
ID TYPE      DEVICE                   SOFT      HARD
 0 wlan      phy0                  blocked unblocked
 1 bluetooth tpacpi_bluetooth_sw unblocked unblocked
 2 wwan      tpacpi_wwan_sw      unblocked unblocked
 3 bluetooth hci0                unblocked unblocked
bencord0@localhost ~/src/xmm7360-pci $ ip r
default dev wwan0 proto static metric 1000 
bencord0@localhost ~/src/xmm7360-pci $ ip addr show dev wwan0
4: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/none 
    inet 10.147.238.58/0 scope global wwan0
       valid_lft forever preferred_lft forever
    inet6 fe80::c479:6009:372:d42c/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
bencord0@localhost ~/src/xmm7360-pci $ ping 1.1.1.1 -c 3
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=54 time=34.7 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=54 time=36.0 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=54 time=43.8 ms

--- 1.1.1.1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 34.687/38.187/43.836/4.032 ms
```

bencord0, 2021-06-01:
On suspend/resume, I get these messages in `dmesg`.

`xmm7360 0000:03:00.0: Failed to ship coalesced frame`

Running `rmmod xmm7360`, `insmod` and then running `open_xdatachannel.py` again renegotiates and brings back networking.

gariac, 2021-06-02:
I'm working on issue 106. Thinkpad 495 on opensuse 15.2. Can you indicate what happens when you do

./scripts/lte.sh up

I get a long number and the window stays busy.

tgxn, 2021-08-02:
@bencord0
Firstly, suspend/resume is known to not work: #1 
You may have to unload/reload the module after suspend/resume to make it work again.

Sounds like the driver's working as intended? You can ping via the LTE connection.

tgxn, 2021-08-02:
I will add info on Gentoo Linux and the `python-dbus` module to `DEVICES.md`

tgxn, 2021-08-02:
Awesome mate! I'll have to add X395 to the `DEVICES.md` list too; with information on `emerge dbus-python`

closed 2021-08-02 by tgxn