xmm7360/xmm7360-pci #108: Works on Lenovo X1 Yoga Gen 3 (20LG) w/ Linux Mint 20

lu0, 2021-06-03:
I've been fiddling around with this for a while now, I though It didn't work out but it turned out I just had to reboot my machine... The connection is displayed in the network manager as "Unknown", but everything runs smoothly.

- Host: Lenovo Thinkpad X1 Yoga Gen 3 (20LG)
- OS: Linux Mint 20 Ulyana
- Kernel: 5.4.0-26-generic
- Carrier: AT&T Mexico

Config file:
```ini
apn=NXTGENPHONE
noresolv=True
dbus=True
BIN_DIR=/usr/local/bin
```
DNS (just in case): `201.130.58.252`, `201.130.58.253`.

Now I just have to figure out how to / if I can receive SMS messages :P

Thanks!!

tgxn, 2021-08-02:
Champion! I'll have to add to `DEVICES.md` Thanks!

tgxn, 2021-08-02:
Thinking potentially create a new issue for SMS? There is some of it covered off in https://github.com/xmm7360/xmm7360-pci/issues/31 though I think that it's likely never going to be supported, unless we make some kind of breakthrough in reversing the windows driver :P

closed 2021-08-02 by tgxn

dammhannes, 2023-01-12:
I can confirm this works fine for ubuntu 22.04.01 LTS but not running at startup!

As a fix I put this line in my /etc/rc.local: 
(cd /*path to the directory*/xmm7360-pci/ && make load && python3 rpc/open_xdatachannel.py)

I am on O2 Germany, hence my config file:
apn=internet
noresolv=True
dbus=True
BIN_DIR=/usr/local/bin

THANK YOU for this amazing work!