xmm7360/xmm7360-pci #110: Lenovo Thinkpad L14: cannot get it to work with ISP Altice/MEO/TMN

tryfail, 2021-06-07:
I have followed the README and I am not able to use the SIM card that I have used successfully in other computers for a few years. The errors are very similar to issue #34 

I have tried running this around ~2 months ago without success and tried it today as well. I believe the output is similar to how it was last time.

The provider is Altice (the one that is based in Portugal). It is a rebranding of MEO/TMN. The configuration that I used in the past with **networkmanager + ericson wwan card** is the following:
- apn: internet
- number: *99#
- username: tmn
- password: tmn

Here is the output of `ip a` after running the command:
```
8: wwan0: <POINTOPOINT,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/none
```

Here are the logs for the execution of the program. I can not make any sense of them

```
xmm7360-pci# python3 rpc/open_xdatachannel.py --apn tmn
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x842b8252
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c020400000192110001010204110001010204207a0e9e'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaMsCpsSetModeIndCb: 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa0204000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x4, 0x0, 0x90, 0x3, 0x1, 0x3, 0x1, 0x0, 0x0
^C
```

Additionally, I am not sure if the following error is due to the one above. When I try to run the program again after using `Ctrl+C` I get this error, when I run `python3 rpc/open_xdatachannel.py --apn tmn`,  if I do not restart first: 
`INFO:root:Attach failed - waiting to see if we just weren't ready`

I have also tried using the wrong APN, but I get the same output. No `.ini` files are in use.

carstencodes, 2021-06-07:
I have a similar issue. I thought I could fix it with elements similar to https://github.com/xmm7360/xmm7360-pci/issues/89#issuecomment-823991902

I'm doing the following:

`$ make`
make -C /lib/modules/5.4.0-73-generic/build M=/home/carsten/dev/xmm7360-pci modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-73-generic'
  CC [M]  /home/carsten/dev/xmm7360-pci/xmm7360.o
  Building modules, stage 2.
  MODPOST 1 modules
  LD [M]  /home/carsten/dev/xmm7360-pci/xmm7360.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-73-generic'
`$ sudo make load`
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
`$ echo "AT+CPIN=\"thisismypin\"" | sudo tee -a /dev/ttyXMM1 `
...
`$ sudo python3 rpc/open_xdatachannel.py --apn internet`
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x5483358d
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c0204000001921100010102041100010102049a98d7a2'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaMsCpsSetModeIndCb: 0x0, 0x0, 0x0, 0x0, 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc02040000000200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002020000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000020400000000551402040000001402040000000000000000000000000000000000000000000000000204000000005565020400000068020400000003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000055820101020400000104020400000003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002040000000055410204000000420204000000010000000000000000000000000000000000000000000000000000000000 [... cut ...]
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x5, 0x0, 0x90, 0x1, 0x0, 0x1, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x2, 0x1, 0x2, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x3, 0x1, 0x3, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x4, 0x1, 0x4, 0x1, 0x0, 0x0

The Device does not show up in Cinnamon Network Connections.

System details:
- Lenovo L14 G1
- Linux Mint 20.1
- Kernel 5.4.0-73-generic

carstencodes, 2021-06-07:
Provider is O2 / Germany btw.

tryfail, 2021-06-08:
@carstencodes  . No success. Thanks anyway.

I did look through that post before. I was already using signed **kernel modules + secure boot** without sucess.

Today I tried the following:
- a different and also PIN-less Altice card. No success and very similar output
- a PIN-less Vodafone card. Tried it with the two national APNs and the wwan0 interface shows up in 'ip a' in state down and without address. Just like my experience with the Altice cards. I am attaching the logs:
  - [vodafone2.txt](https://github.com/xmm7360/xmm7360-pci/files/6612687/vodafone2.txt)
  - [vodafone1.txt](https://github.com/xmm7360/xmm7360-pci/files/6612688/vodafone1.txt)

At this point should I be considering dropping secure boot? Or just trying a fresh Xubuntu 20.04 without any customizations?

Thanks

carstencodes, 2021-06-08:
Apparantly I don't know. I'm running into similar issues on the similar hardware. So I thought, why not joining forces.

tgxn, 2021-08-02:
Just brushing over this thread, is this not related to #109 PAP/CHAP Authentication not being supported?

carstencodes, 2021-08-02:
@tgxn I think so, but don't you mean #62 as the related issue rather then 109?

carstencodes, 2021-08-02:
I'll try to remove the PIN later...

carstencodes, 2021-08-02:
@tgxn I removed the PIN. Now it works. I had to revise the APN name, but now it works using lte.sh up/down.

tgxn, 2021-08-03:
@carstencodes Haha! Yeah I totally got that mixed up! Have updated comment.

@tryfail Not sure if you've made any more progress, were you able to get an LTE connection active on the device yet?

tryfail, 2021-08-03:
> @carstencodes Haha! Yeah I totally got that mixed up! Have updated comment.
> I'll also tag this with `device-support` and add your findings to the `Devices` document.
> 
> @tryfail Not sure if you've made any more progress, were you able to get an LTE connection active on the device yet?

I have just pulled-checkout master and the output is the same. I am using the same laptop as previously. Do note that:
- this is most likely not an LTE capable SIM card. It has a 3G plan that has always been enough for me considering the very decent UMTS/HSPA+ coverage. For example, I can use it on the highway and in faster trains
- the Altice/MEO/PT manufacturer does not mention CHAP or PAP requirements. Not sure if collides with the fact that a password is required
  - national Vodafone does mention requiring CHAP
- no PIN is in use. Never was
- I have tried using the lte.sh, but the behaviour is the same. I was already using the same commands

Out of ideas at this point. I have not been capable of troubleshooting this. My mind just goes numb after reading other issues on github for this project, since I am just running into generic errors. WWAN on laptops is something I usually only touch if there is a known working recipe. I would have installed a different card, but Lenovo will not
 allow it on the L14 Gen1 (maybe because L14 G1 + Fiboxcom is certified for Ubuntu :>)

closed 2021-08-03 by tryfail