xmm7360/xmm7360-pci #111: can't build on carbon x1 gen 7 and fedora 33

zeropoint46, 2021-06-16:
trying to build on fedora using the scirpt and I get the following error:

```make -C /lib/modules/5.12.10-200.fc33.x86_64/build M=/home/jpsergent/Workspace/xmm7360-pci modules
make[1]: Entering directory '/usr/src/kernels/5.12.10-200.fc33.x86_64'
  CC [M]  /home/jpsergent/Workspace/xmm7360-pci/xmm7360.o
/home/jpsergent/Workspace/xmm7360-pci/xmm7360.c: In function ‘xmm7360_create_tty’:
/home/jpsergent/Workspace/xmm7360-pci/xmm7360.c:1332:10: error: ‘struct tty_port’ has no member named ‘low_latency’
 1332 |  qp->port.low_latency = 1;
      |          ^
make[2]: *** [scripts/Makefile.build:271: /home/jpsergent/Workspace/xmm7360-pci/xmm7360.o] Error 1
make[1]: *** [Makefile:1852: /home/jpsergent/Workspace/xmm7360-pci] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.12.10-200.fc33.x86_64'
make: *** [Makefile:9: default] Error 2
sudo /sbin/rmmod xmm7360
rmmod: ERROR: Module xmm7360 is not currently loaded
make: [Makefile:18: load] Error 1 (ignored)
sudo /sbin/insmod xmm7360.ko
insmod: ERROR: could not load module xmm7360.ko: No such file or directory
make: *** [Makefile:19: load] Error 1```

RyanShaw-3083, 2021-06-16:
Hi @zeropoint46, This (#100) issue should solve your problem. You can try to comment out the code at line 1332.

zeropoint46, 2021-06-17:
@RyanShaw-3083 thanks, I found that after I posted this and it worked. Thanks.

closed 2021-06-17 by zeropoint46