xmm7360/xmm7360-pci #112: issue connecting with open_xdatachannel

zeropoint46, 2021-06-16:
trying to connect with my modem and getting the following error:

```
ERROR:root:Attach failed again, giving up
```
saw another topic on tihs but it is not simpin related. I don't have a pin enabled on my sim. not sure when else to go. Additionally while monitoring minicom when I run the script, I get "+GTURCREADY: 0" in the output. not sure if thats good or bad.

dominikpegler, 2021-07-08:
Experiencing the same since today on Thinkpad Carbon X1 6G with Manjaro 21.07. Yesterday it worked and I cant figure out why it stopped.

jonpolak, 2021-08-01:
same issue. Thinkpad T14s.

tgxn, 2021-08-02:
I've noticed similar issues seemingly related to #1 

Can you see if a full unload+reload and then connecting again works for you?

dominikpegler, 2021-08-02:
> Experiencing the same since today on Thinkpad Carbon X1 6G with Manjaro 21.07. Yesterday it worked and I cant figure out why it stopped.

I found the issue to be the SIM lock -> https://github.com/xmm7360/xmm7360-pci/issues/116#issuecomment-879394391