xmm7360/xmm7360-pci #114: error in X1 Nano

faffa42, 2021-06-25:
Hi trying to run it on a brand new Thinkpad Nano, but I get this error
`~/tmp/xmm7360-pci$ sudo python3 rpc/open_xdatachannel.py --apn internet.wind
Traceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 42, in <module>
    r = rpc.XMMRPC()
  File "/home/faffa/tmp/xmm7360-pci/rpc/rpc.py", line 19, in __init__
    self.fp = os.open(path, os.O_RDWR | os.O_SYNC)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/xmm0/rpc'
`
On Ubuntu 20.04, any idea?

t11c, 2021-07-31:
I had a similiar error on Ubuntu 21.04,  disconnecting any wifi networks solved it for me.

tgxn, 2021-08-02:
I've also seen this error after suspend/resume cycle as in #1.

Please try:
- Ensure WiFi Is disabled.
- Unload module + restart system
- make/make load driver, test connection.

tgxn, 2021-08-02:
X1 Nano also reported working here: https://github.com/xmm7360/xmm7360-pci/issues/125