xmm7360/xmm7360-pci #115: Works on Lenovo X1 Yoga Gen 4 | Arch Linux

3ndlos, 2021-06-25:
Had to remove line 1332 ( qp->port.low_latency = 1; ) from 'xmm7360.c' to run lte.sh setup but now everything works like a charm.
Connection is displayed as "Unknown" but it doesn't seem to have any effect.

>  Host: Lenovo Thinkpad X1 Yoga Gen 4
>  OS: Arch Linux
>  Kernel: 5.12.12-arch1-1
>  Carrier: WinSIM (O2 | DE)

Copied xmm7360.ini.sample to xmm7360.ini and just changed apn to "internet".
No SIM-PIN.
Nothing else needed.

plajjan, 2021-06-29:
+1 on this. It works on X1 Yoga Gen4 for me running Debian unstable with kernel 5.10.0-3-amd64.

I did not have to remove line 1332. Might be related to what the mobile operator supports perhaps?

I just copied xmm7360.ini and changed the apn to 4g.tele2.se, which is for Tele2 in Sweden, though I'm pretty sure they do APN override so any value would work.

tgxn, 2021-08-02:
Hey, The `low_latency` issue should be resolved after #117 was merged.

Good to hear it's working, I'll add your device/OS to `DEVICES.md`

closed 2021-08-02 by tgxn