xmm7360/xmm7360-pci #116: Lenovo T480s + manjaro 21.0.7

ItanVazlak, 2021-07-03:
Good day. I was able to run this driver on Lenovo T480s + manjaro 21.0.7

To do this, enter the following commands on the command line:

**mkdir ~ / tmp /
cd ~ / tmp /
sudo pacman -S git
sudo pacman -S linux-headers**

This package was installed on the advice of the user Lenovo t14. After installing the package, you need to select the version corresponding to your kernel version. In my case 5.10

**sudo pacman -S python-pyroute2**

For Linux Ubuntu, the python3-pyroute2 package is used. This package is not available on Manjaro. There is only python-pyroute2. We install this package.

**sudo pacman -Sy base-devel**

The base-devel package is a replacement for the build-essential package for Ubuntu

**sudo pacman -S python-configargparse**

For Linux Ubuntu, the python3-configargparse package is used. This package is not available on Manjaro. There is only python-configargparse. We install this package.

**git clone https://github.com/xmm7360/xmm7360-pci.git
cp xmm7360.ini.sample**

Next, you need to edit the xmm7360.ini file. In my case, it was enough only to register the provider's APN.

_! # driver config
apn = internet.beeline.ru

# ip-fetch-timeout = 1
# nodefaultroute = True
# metric = 1000

# uncomment to not add DNS values ​​to /etc/resolv.conf
# noresolv = True

# used to activate NetworkManager integration
# dbus = True

# Setup script config
BIN_DIR = / usr / local / bin_

**cd xmm7360-pci
make && make load**

The first time the command "make && make load" will fail. I must repeat the command

**make && make load
sudo python rpc / open_xdatachannel.py**

when executing this command, the following error occurs:

  _File "/home/ukulele/tmp/xmm7360-pci/rpc/rpc.py", line 209, in unpack_unknown
    raise ValueError ("unknown type 0x% x"% t)
ValueError: unknown type 0x5_

**sudo ip link set wwan0 up**

after that the internet started working.

Features of LTE

1.The LTE connection is maintained until the router is rebooted. After rebooting, you have to repeat the following commands in the terminal:

**cd ~ / tmp /
cd xmm7360-pci
make && make load
make && make load
sudo python rpc / open_xdatachannel.py
sudo ip link set wwan0 up**

2. Modem Manager GUI does not see this LTE modem

3. The new interface is visible only in the terminal using the command

**ip addr**

dominikpegler, 2021-07-08:
This was very helpful for yesterday. I am on a 

- Thinkpad X1 Carbon 6G
- Manjaro 21.07. (Kernel 5.11.22-2-MANJARO)
- latest version of this repo

Yesterday it worked exactly as described above. Today it stopped working and I am having a hard figuring out why, because I am not aware that anything has changed since then.

That's what I get when I am running `sudo python ./rpc/open_xdatachannel.py`

`ERROR:root:Attach failed again, giving up`

ItanVazlak, 2021-07-08:
Good day.

Unfortunately, I am not good at Linux. Approximately 1-2 points out of 10 possible. When I wrote the post above, I hoped the author of the driver would help. There is clearly an error somewhere, but I don't know where. I am ready to provide access on my laptop to my system.

The following helps me:
1.reboot laptop
2.turn off WiFi
3. Execute all commands in this sequence
cd ~/tmp/
cd xmm7360-pci
make && make load
make && make load
sudo python rpc / open_xdatachannel.py
sudo ip link set wwan0 up

dominikpegler, 2021-07-13:
thank you for the update.

unfortunately, updating the linux-headers didn't change anything.

dominikpegler, 2021-07-13:
I got it working again.

The problem was with the SIM card being locked. I had to unlock it.

Enter minicom: `$ minicom -D /dev/ttyXMM0`

Check if SIM card is locked: `AT+CPIN?`
=> if it answers with "+CPIN: SIM PIN" then it's locked, if "+CPIN: READY" then it's unlocked

Unlocking it (my pin here is 1234):`AT+CPIN="1234"`
=> It should answer with "OK" and if you issue the previous command it should say "+CPIN: READY"

Turning off SIM lock permanently (in case you want to):`AT+CLCK="SC",0,"1234"`
=> It should answer with "OK"

tgxn, 2021-08-02:
Hey @dominikpegler  and thanks for your findings. I hope that with your lat4est update you've managed to get it working.

Assigned to me to add `G6/Manjaro` to `DEVICES.md`

tgxn, 2021-08-02:
@ItanVazlak 
I updated your original comment to include formatting that GitHub understands, I hope you don't mind. I think it could help future people.

There is a known issue that is why you must restart the laptop / reload driver Issue #1 has more info on it but no solution yet.

I'll add Manjaro and T480 to the `DEVICES.md` document as well, thanks for your testing!

closed 2021-08-02 by tgxn

samtech09, 2021-08-13:
> Conclusion: With an unlocked SIM (see above) and below steps after a reboot it's working on my system:
> 
> make && make load
> make && make load 
> sudo python rpc/open_xdatachannel.py 
> sudo ip link set wwan0 up**
> 

Working well on Thinkpad X1C 7th Gen.