xmm7360/xmm7360-pci #118: Syntax error in rpc/open_xdatachannel.py

SergeyDjam, 2021-07-07:
Hello. 
```
sudo python3 rpc/open_xdatachannel.py
  File "rpc/open_xdatachannel.py", line 83
    logging.info(f"IP address couldn't be fetched, waiting {interval} seconds")
                                                                             ^
SyntaxError: invalid syntax

```

HP EliteBook 830 G6, Linux 5.10.31-generic-1rosa-x86_64, ROSA Linux r11.1, 

steps
1. git clone
2. make & make load
3. sudo pip install --user pyroute2 ConfigArgParse
4. sudo python3 rpc/open_xdatachannel.py --apn internet.beeline.ru

SergeyDjam, 2021-07-07:
I am commented 93 string and start, but no work

```
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
^CTraceback (most recent call last):
  File "rpc/open_xdatachannel.py", line 68, in <module>
    r.pump()
  File "/var/disk/admin/rpmbuild/BUILD/xmm7360-pci/rpc/rpc.py", line 27, in pump
    message = os.read(self.fp, 131072)
KeyboardInterrupt
```

SergeyDjam, 2021-07-07:
> 3b:00.0 Wireless controller [0d40]: Intel Corporation Device 7360 (rev 01)

dearfl, 2021-07-09:
f-string requires python>=3.6

closed 2021-07-19 by SergeyDjam