xmm7360/xmm7360-pci #121: HP EliteBook 830 G6 don't work

SergeyDjam, 2021-07-19:
uname -a
Linux nb 5.10.47-generic-1rosa-x86_64 #1 SMP Wed Jun 30 22:30:02 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

lsmod | grep xm
xmm7360                24576  1

lspci
3b:00.0 Wireless controller [0d40]: Intel Corporation Device 7360 (rev 01)

sudo /usr/libexec/python3.8 rpc/open_xdatachannel.py --apn internet.beeline.ru
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x32ff, 0x25, 0x0, 0x90, 0x6, 0x1, 0x6, 0x1, 0x3, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0xff, 0x27, 0xb, 0x91, 0x7, 0x1, 0x7, 0x1, 0x0, 0x0
unsolicited: UtaMsSimTkTerminalResponseIndCb: 0x0, 0x0, 0x2004, 0x26, 0x0, 0x90, 0x8, 0x1, 0x8, 0x1, 0x0, 0x0

ifconfig wwan0
wwan0: flags=4240<POINTOPOINT,NOARP,MULTICAST>  mtu 1500
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

how i can fix this problem?

tgxn, 2021-08-02:
Few things to try:
- Make sure WiFi is diabled.
- Make sure no SIM PIN on SIM.
- Rebuild+reload module
- Re-try `open_xdatachannel`
- Include full log outputs for above.

so-rose, 2022-10-23:
I seem to be getting the exact same error mesage on my X1 Yoga Gen 3, though without the `unsolicited: UtaMsSimTkTerminalResponseIndCb`

```
$ sudo ./rpc/open_xdatachannel.py --apn Internet
...
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
```

What does the error message mean?

# Checks
- I tested the SIM card in my phone and grabbed the APN name from there too. No PIN as far as I can see.
- I turned off WiFi via ConnMan. So it should be good to go.
- I've recompiled the module from latest master (I've automated this via DKMS).

`uname -r`: `5.18.0-0.deb11.4-amd64`
`uname -a`: `Linux <> 5.18.0-0.deb11.4-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.16-1~bpo11+1 (2022-08-12) x86_64 GNU/Linux`

# Complete Log
```
$ sudo ./rpc/open_xdatachannel.py --apn Internet
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x8bc1443
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c02040000019211000101020411000101020412ec13e3'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000020200000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000000204000000005514020400000014020400000000000000000000000000000000000000000000000002040000000055650204000000680204000000030000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000558201010204000001040204000000030000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...] 0000000000000000000000000000000000000000000000000000000000000000000000000000000000020103020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x0
RPC executing UtaMsNetAttachReq
b'4500000002040000004502040000005c110001010204110001010201000204000000000204000000000204000000000204000000000202ffff0202ffff020400000000020400000000'
async_ack: 0x0
response: 0x0, 0xffffffff, 0x5, 0x5, 0x12, 0x0, 0x0, 0x0, 0x0, 0x0, 0x5, 0x0, 0x5, 0x12, 0x0
INFO:root:Attach failed - waiting to see if we just weren't ready
```