xmm7360/xmm7360-pci #122: Don't work on Lenovo X1 Carbon gen 7 with PoPOS! 21.04

Yetiicom, 2021-07-21:
Machine: Lenov X1 Carbon gen 7
OS: PoP OS! 21.04
Description: 
After the xmm7360 repository was cloned I followed the installation guide and got an error. 

```
$ cp xmm7360.ini.sample xmm7360.ini 
$ nano xmm7360.ini
$ sudo ./scripts/lte.sh setup
$ lte up
```

```
This script must be run as root, elevating!
lte.sh: manage xmm7360-pci
APN: internet
Script: /home/xxx/Downloads/xmm7360-pci/scripts/lte.sh; Link: /usr/local/bin

bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
Traceback (most recent call last):
  File "/home/xxx/Downloads/xmm7360-pci/scripts/../rpc/open_xdatachannel.py", line 68, in <module>
    r.pump()
  File "/home/xxx/Downloads/xmm7360-pci/rpc/rpc.py", line 28, in pump
    resp = self.handle_message(message)
  File "/home/xxx/Downloads/xmm7360-pci/rpc/rpc.py", line 93, in handle_message
    content = unpack_unknown(body)
  File "/home/xxx/Downloads/xmm7360-pci/rpc/rpc.py", line 209, in unpack_unknown
    raise ValueError("unknown type 0x%x" % t)
ValueError: unknown type 0x5
```

tgxn, 2021-08-02:
This is a new error I've not seen before. Requires further investigation.

tgxn, 2021-08-02:
This guy got the error then it started working: https://github.com/xmm7360/xmm7360-pci/issues/116 though he's on Manjaro.

Yetiicom, 2021-08-03:
> This guy got the error then it started working: #116 though he's on Manjaro.

I have read thread #116, but it seems that author have a problem with PIN set on SIM card. 
In my case I don't have PIN on my SIM card. The card configuration was checked on mobile phone and no PIN is required to use the card.

Oualid33, 2022-08-02:
I got the same error message but for `UtaMsNetAttachReq; Hereafter the output of the script:
`bringing wwan0 up!
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x0, 0x2
FCC lock: state 0 mode 2
RPC executing CsiFccLockGenChallengeReq
b'1c00000002040000001c02040000019011000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x4ea69cfc
RPC executing CsiFccLockVerChallengeReq
b'1c00000002040000001c02040000019211000101020411000101020413150703'
async_ack: 0x0
response: 0x1
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaMsSsLcsCapabilitiesIndCb: 0x0, 0x1, 0x10, 0x10, 0x10, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x0, 0x1, 0x1, 0x1, 0x1, 0x0
unsolicited: UtaMsNetExtBandStatusIndCb: 0x0, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x0, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x32, 0x5b, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x100, 0x3a0f18df, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x100, 0x3a0f18df, 0x32, 0x32, 0x32, 0x33, 0x32, 0x35, 0x32, 0x36, 0x32, 0x39, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x5, 0x35, 0x32, 0x35, 0x33, 0x35, 0x35, 0x35, 0x36, 0x35, 0x39, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x32, 0x4b, 0x5, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x12, 0x0, 0x0
unsolicited: UtaMsNetFdConfigIndCb: 0x0, 0x1, 0xff, 0x0, 0x0, 0x0, 0x5, 0x4, 0x0
unsolicited: UtaMsCpsSetModeIndCb: 0x0, 0x0, 0x0, 0x1, 0x0
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x3, 0x1, 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc020400000002000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000020200000204000000000204000000000204000000000204 [... cut ...]
    attach = r.execute('UtaMsNetAttachReq',
  File "/home/oualid/tmp/xmm7360-pci/rpc/rpc.py", line 69, in execute
    resp = self.pump()
  File "/home/oualid/tmp/xmm7360-pci/rpc/rpc.py", line 28, in pump
    resp = self.handle_message(message)
  File "/home/oualid/tmp/xmm7360-pci/rpc/rpc.py", line 93, in handle_message
    content = unpack_unknown(body)
  File "/home/oualid/tmp/xmm7360-pci/rpc/rpc.py", line 209, in unpack_unknown
    raise ValueError("unknown type 0x%x" % t)
ValueError: unknown type 0x5
`

GreatEmerald, 2023-06-12:
I am getting the same error every time I try to run open_xdatachannel.py. My output is this:
```
> sudo python3 rpc/open_xdatachannel.py 
RPC executing UtaMsSmsInit
b'1600000002040000001602040000003011000100020400000000'
response: 0x0
RPC executing UtaMsCbsInit
b'1600000002040000001602040000002511000100020400000000'
response: 0x0
RPC executing UtaMsNetOpen
b'1600000002040000001602040000005311000100020400000000'
response: 0x0
RPC executing UtaMsCallCsInit
b'1600000002040000001602040000002411000100020400000000'
unsolicited: UtaMsNetGprsClassIndCb: 0x0, 0x3, 0x1, 0x0
response: 0x0
RPC executing UtaMsCallPsInitialize
b'1600000002040000001602040000003a11000100020400000000'
response: 0x0
RPC executing UtaMsSsInit
b'1600000002040000001602040000002611000100020400000000'
response: 0x0
RPC executing UtaMsSimOpenReq
b'1600000002040000001602040000000111000100020400000000'
response: 0x0
RPC executing CsiFccLockQueryReq
b'1c00000002040000001c02040000018e11000101020411000101020400000000'
async_ack: 0x0
response: 0x0, 0x1, 0x0
FCC lock: state 1 mode 0
RPC executing UtaModeSetReq
b'2200000002040000002202040000012f1100010002040000000002040000000f020400000001'
response: 0x0
unsolicited: UtaModeSetRspCb: 0x1, 0x0, 0xf
RPC executing UtaMsCallPsAttachApnConfigReq
b'e11300000204000013e10204000001af110001010204110001010201005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000202000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000002040000000055140204000000140204000000000000000000000000000000000000000000000000020400000000556502040000006802040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005582010102040000010402040000000300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000204000000005541020400000042020400000001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000554102040000004102040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005581fa0204000000fa020400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000201005581fa0204000000fc0204000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
  File "/home/dainius/Documents/xmm7360-pci/rpc/rpc.py", line 28, in pump
    resp = self.handle_message(message)
  File "/home/dainius/Documents/xmm7360-pci/rpc/rpc.py", line 93, in handle_message
    content = unpack_unknown(body)
  File "/home/dainius/Documents/xmm7360-pci/rpc/rpc.py", line 209, in unpack_unknown
    raise ValueError("unknown type 0x%x" % t)
ValueError: unknown type 0x5
```

That said, I don't have a SIM card inserted (I don't have any SIM cards with a data plan - yet). Nevertheless, the script should not crash because of that...

Oualid33, 2024-08-27:
Having the same issue, I removed the SIM card, inserted into phone, then reinserted back in my T14.
I updated Ubuntu 22.04 LTS packages (via WiFi), then repeated the same compilation/setup procedure:

$ sudo rmmod iosm && cd /tmp/xmm7360-pci && make clean && sudo make && sudo load && sudo lte up

and I worked !!
Don't know why ?! Without changing anything else.